john30/ebusd-esp #4: ebusd-esp stuck in TX loop ?

ChrisP-Git, 2018-01-30:
Hi,

I have the new ebus2 adaptor since this week-end, and had a very strange issue twice since then:

I'm using the new OpenHab2 ebus binding, with much more success than with esera ethernet coupler. 

After some time, I remark that the boiler stopped being regulated, and just start to heat very strongly, and the VRC470 screen complained about lost ebus communication.

- On the eBus adaptor, I can see the red LED continously blinking, one or two time pe.cond, yellow led is lit and green LED seems to blink as usual, but less visible than the RED, so I'm not so sure about this.
- On the ebusd-esp web page, I can see than the connection is not active anymore.
- On OpenHab side, the binding is disconnected, and logs does not show specific error message, just that communication stopped with the adaptor.

I guess the ebusd is stuck in, a loop, spaming the ebus bus, preventing the VRC470 to effectively control the boiler. As soon as I disconnect the adaptor from the ebus, boiler come back to standard behavior, as well as VRC470.

Since the connection is lost with the openHAB binding, the origin of the spam shouldn't be the binding itself.

I have then many questions:

- So what could explain this condition, especially why is the ebusd-esp continusly sending things while being disconnected on TCP/UDP client side ?
- What could be done to troubleshoot this ? 
- Is there a way to get a log of what the ebusd-esp is sending or receiving ? A special debug web page would help a bit.
- I'm a bit reluctant to let it connected without supervision for now :/ Is there a way to prevent the boiler to heat up too much in case of such event ? I am wondering if lowering the manual heat poti, that is supposed to be set at max when connected to a VRC could be lowered for that.

ChrisP-Git, 2018-02-01:
I just had it a third time, with very limited traffic from the openHab binding, (only 2 data was requested at a pace of one request every 600 seconds). 
Then again, the red led was blinking contunously, VRC470 lost communication, and the ebusd-esp TCP session was disconnected.
But as soon as I connected directly to the ebusd-esp TCP port using RealTerm, everything went back to normal, and traffic was properlly displayed within Real Term.

john30, 2018-02-03:
first of all: which version of ebusd-esp are you using?
second: how is it configured? please post a summary of the settings (without your Wifi password of course)

ChrisP-Git, 2018-02-03:
Hi John,

so Build is 20171230, it was preloaded with the ebus2 adaptor.
ebusd TCP/UDP mode = TCP, Port = 9999
RX+TX PINs = direct RX + TX (GPIO3+1)
LED PINs RX disabled, TX disabled

However the behavior is not during standard openhab operation but observed once the TCP session is closed, having ebusd connected=no displayed on ebusd-esp management page, so I do not think openhab behavior have an impact in this very specific situation.

john30, 2018-02-03:
ok, and why did openhab close the session? and did it close it properly? you'd have to do a tcpdump to see that

ChrisP-Git, 2018-02-03:
That's a good question, I do not know which side close the session, or it can be Wifi related as well.

However, is it expected that the ser2net feature continue to send data (if that's what the blinking red LED is showing) once the session is closed for whatever root cause ?

ChrisP-Git, 2018-02-03:
I mean, I'm not ranting here, but I would expect there to just stop communication within the ebus side in such a situation, and it's not what I'm observing.

john30, 2018-02-03:
of course not. once the tcp connection enters any other state than ESTABLISHED, no more data is sent to ebus. Can you try to reproduce the behaviour and let tcpdump run on the port ony your openhab server?
Another thing to look at might be your wemos power supply. If that is not stable enough, then weird behaviour might be seen. It could be the same here. Can you try with a different power supply?

ChrisP-Git, 2018-02-05:
Power Supply is a a thing I have to investigate, as I am using the PC USB port, which should be limited at 500 mA.
Actually reproducing the issue is the harder thing, as it took 12 hours the first time to have the issue. I have to check how I can make a tcpdump that long, and I want to use a digtal oscilosope as well to check the ebus state once the issue is triggered.

ChrisP-Git, 2018-02-28:
Ok, I have some additional hints after some more test. Actually the issue is really random, it is pretty hard to reproduce.

So it happened once when nothing was connected to the ser2net port, I hadn't time to reconfigure anything to connect to it, so the data sent to the converter seems not related.

More interesting, once in the dead loop, using arduino ide, I connected the serial monitor to the serial port. and seen the configuration menu sent continuously, with a line of random garbage characters printed after the line "Enter your choice: ", then the menu printed again at a very fast rate and looping continuously.

ChrisP-Git, 2018-02-28:
So I was able to make some capture with a scope. When working properlly, this is what it looks like when the SYN symbol is sent:
[image]

ChrisP-Git, 2018-02-28:
And this is what it looks like, with the same time resolution:
[image]

ChrisP-Git, 2018-02-28:
Zooming out to 50 ms/div, you can see this strange traffic sent at regular basis:

[image]

john30, 2018-03-01:
this all sounds like a power issue. once the wemos consumes too much power, it will reboot and then somehow enter the endless loop (which it shouldn't). so please first replace the power supply

ChrisP-Git, 2018-03-27:
Hi, 

so I finally found the good opportunity to be able to let the device connected for a long time while having the ability to react in case of issue. 

So I used a proper 1000mA USB PSU, and the device is working perfectly fine for more that 10 days in a row now.

I think some specific and explicit warning should be provided regarding PSU, because when the problem arise, the boiler can go to full throttle non stop, without any regulation. 
This can damage the boiler, or even worse !!

john30, 2018-03-27:
sure. you attach a self-made device to your heating. so what's the point here?

closed 2018-03-27 by john30