john30/ebusd-esp #8: Feature request: Toggle GPIO pin from UI

FKuypers, 2018-09-10:
I am trying to control a Wolf CWL Excellent 400 with this setup. I am able to control the fans using EBUSD, but since I also have a controlunit attached to the EBUS, it basically resets the fan speed every 30 seconds to it's original value.

My idea is now to disconnect the controlunit from the EBUS during the time I want to control the CWL using EBUSD. Once I want the automatic program to continue, I simply reconnect the EBUS to the controlunit. 

For this I would use a IRF520 connected to a GPIO of the ESP. Hence it would be great, if I would switch a GPIO through a call to the web interface of the EBUS-ESP.

hvorragend, 2018-09-10:
I have the same problem. My wall remote control is the master device and makes a change every few seconds. 
But: If I would interrupt the bus to this remote, I would have no display in this remote. 

In addition, I have connected a switch on the ventilation system, with which I can also override the fan speeds.
However, I see on the eBus that the resets are constantly being transmitted.

FKuypers, 2018-09-10:
I have seen that you can also connect the "old style" fan speed selector on the RJ11 connection and there is a setup guide on how to emulate this with Homematic for example - but does this selector have priority over the EBUS connected remote? 

I don't mind not having a display in the remote, since the usage typically is to dial down the fans at night from my phone, since someone set it to "high" before and I want to avoid having to go to the remote to set it back to "automatic" - which sets the fans to low at night anyway...

Maybe there is a way to actually talk to the remote using EBUS - and tell the remote to tell the system the setting I want....

john30, 2018-09-16:
feature was added, see readme

closed 2018-09-16 by john30