john30/ebusd-esp #10: The last dropped *.bin is not working

sergs333, 2018-09-19:
Hi,

Currently hosted *.bin is continuously rebooting throwing dump in COM port. I've succsseed flashig last release from Dec 2017.

hvorragend, 2018-09-19:
For me it is not possible to save the settings. After saving, "tcp enhanced" reoccurs.

sergs333, 2018-09-19:
Try fill flash with 0x00 and then reflash the unit. It could help. Here is article how to do this http://www.pratikpanda.com/completely-format-erase-esp8266-flash-memory/. But for me it didn't help.

john30, 2018-09-19:
@sergs333 what device do you flash it onto? please also post the beginning of the output to the COM port after pressing the reset button.

sergs333, 2018-09-19:
NodeMCU
`=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2018.09.19 16:51:54 =~=~=~=~=~=~=~=~=~=~=~=

Welcome to eBUS adapter 2.0, build 20180916
Configured as WIFI access point EBUS without password.
For configuration with web browser, connect to this WIFI and open http://192.168.4.1/
Entering configuration mode.
Chip ID: 00FDD67C

Exception (28):
epc1=0x402121f0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff1550 end: 3fff1990 offset: 01a0

>>>stack>>>
3fff16f0:  40210c74 3fff17f0 3fff178d 4020cf98  
3fff1700:  00000006 3ffef920 3fff24ac 4021534b  
3fff1710:  3ffe8364 0000000f 00000010 3ffeab32  
3fff1720:  3fff1793 3ffe9df4 00000000 4010020c  
3fff1730:  3ffef95c 3ffef97c 00000000 3ffe9e00  
3fff1740:  00000000 3ffe8364 3fff17f0 402110d4  
3fff1750:  00000000 ffffffff 00000000 00000000  
3fff1760:  00000006 0000000a 00302073 40104b1d  
3fff1770:  4000050c 3fff1870 3fff1812 4020cf98  
3fff1780:  402333d1 00000030 0000001e 444446ff  
3fff1790:  00433736 00005801 00000000 3fff3364  
3fff17a0:  000000ea 3fff34a8 00000000 00000000  
3fff17b0:  3fff1920 3fff18f0 0000000c 00000008  
3fff17c0:  3ffe9dff 0000000a 3fff0968 00000030  
3fff17d0:  00000010 00000010 00000000 3fff0968  
3fff17e0:  3fff08a8 3ffe8364 00000040 4020eab5  
3fff17f0:  3fff18aa 40242080 00000035 ffff0208  
3fff1800:  3fff18a0 0000003f 3fff347c 4020af7b  
3fff1810:  3fff330c 00000004 3fff347c 4020afc7  
3fff1820:  40206fbc 3fff18cc 3fff347c 4020aff9  
3fff1830:  3fff1924 3fff06c4 3fff347c 4020b100  
3fff1840:  3ffea802 00000001 3fff3454 4020b128  
3fff1850:  00000000 00640400 0104a8c0 402091d1  
3fff1860:  3fff1bc8 00000008 3fff08a8 4020eaf8  
3fff1870:  3fff1920 3fff18f0 00000008 4020c074  
3fff1880:  3fff1bc8 0000021e 0000021e 00000012  
3fff1890:  3fff08a8 00000008 3fff18a0 4020acac <
3fff18a0:  74736f48 656d616e 4630203a 37364444  
3fff18b0:  40000a43 3fff06c4 3fff06c4 40233258  
3fff18c0:  4020bce8 3fff06c4 3fff34ac 4020805e  
3fff18d0:  00000000 4020bcfc 3fff1924 40100690  
3fff18e0:  3fff1920 3fff18f0 00000008 40208b2c  
3fff18f0:  00000001 4020bcfc 00000000 0000007e  
3fff1900:  6000001c 00000012 3fff086c 00000002  
3fff1910:  3ffe9df4 00000000 3fff18f0 3fff1920  
3fff1920:  40207433 3fff08a8 3fff1940 3fff0968  
3fff1930:  00000001 3fff086c 3fff08a8 40205af4  
3fff1940:  00000000 00000000 00000000 3ffea810  
3fff1950:  0104a8c0 feefeffe feefeffe feefeffe  
3fff1960:  feefeffe feefeffe feefeffe 3fff0968  
3fff1970:  3fffdad0 00000000 3fff0961 4020ba38  
3fff1980:  feefeffe feefeffe 3fff0970 40100700  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
`

john30, 2018-09-22:
can't reproduce this issue. did you flash it correctly?

sergs333, 2018-09-24:
I think I did it correctly.  I even flash Node MCU with 0x00 before flashing you bin. Also the previouse version for Dec 2017 is working.  The flashing mechanism I've used is the same.

john30, 2018-09-25:
weird. again the question: which board do you use exactly?
what makes me wonder is the "NodeMCU" line in your output, I've never seen that before...

szczukot, 2018-09-30:
I use Wemos D1 mini. All bin from this year not working for me. Last from 2017 works ok.
I flash esp8266, blue led is blinking, I see EBUS wifi, but I can't connect to it form phones. All time connected/disconnected..
Also from putty - can't connect.
Version 12.2017 work properly.

john30, 2018-10-03:
maybe you need to reflash the Wemos completely, I had something similar quite some time ago with a factory shipped Wemos and I had to flash like all areas completely from scratch. I think I did it according to [this tutorial](https://wiki.wemos.cc/tutorials:get_started:revert_to_at_firmware) by then

szczukot, 2018-10-03:
Why is all versions from 2017 work ?

matzep2002, 2018-10-19:
I Follow the instction step by step, and have the same Problem. When i make a Flash witch ESPEasy all works fine. But After the Flash witch ebusd esp dont work.
Wifi coneced and disconeced. The HTTP Webpage is Not possible to open. 
Over Putty its Not Open the configuration Page.
I restore it to factory settings. After that same Problem.
But witch ESPEasy again the Wemos works.

Any Solution? 

Thanks

john30, 2018-10-20:
please try again with the latest binary. it was built using Arduino IDE instead of PlatformIO again

sergs333, 2018-11-24:
I've just try latest one and it didn't help :( I see the same behavior as I described above

sergs333, 2018-11-24:
Hm.... I've found wemos device,  and the lite version has problem above, the standard one works.
The only difference between node-mcu and wemos is in the light sensor and 3 color diode. Maybe the root cause is there....

atideath, 2019-02-28:
> I Follow the instction step by step, and have the same Problem. When i make a Flash witch ESPEasy all works fine. But After the Flash witch ebusd esp dont work.
> Wifi coneced and disconeced. The HTTP [...]

I have a same issue with this adapter. How can I sold the problem? ESP is working about 2 hours and stopped. Not reach website but I can ping the device.

john30, 2019-02-28:
which version are you using?
ebusd-esp: ?
Wemos: ?

john30, 2019-04-06:
closed due to inactivity

closed 2019-04-06 by john30