john30/ebusd-esp #11: esp crashes often

Xeno22, 2018-09-30:
Hey John,

Against which version of esp8266 library is this firmware linked? Is it 1.7.1 or later?

I just gave your firmware a try yesterday (I want to passive read my eBUS), but the ESP crashes very often.

Also it is annoying that the firmware configuration has no option to freely use any digital input pin. Also I would like to disable TX completely, since I do not want to write to the eBUS.

My ESP board is a cheap chinesium one: 
NodeMcu Lua ESP8266 ESP-12E CH340G

john30, 2018-09-30:
lovely comment, that really gives me a kick to continue development.
anyway:
first, the firmware is linked against Arduino core 2.4.2.
second, you should check your power supply first since we found out that this very often results in esp crashes.
thrid, if you don't want to write to the bus, simply disconnect TX or run ebusd in readonly mode.
fourth, this firmware is designed to run on Wemos D1 mini primarily as stated in the readme, no support for other hardware at this time and no plans to do so.

Xeno22, 2018-09-30:
I always do lovely comments ;)

Thanks for your fast response and your software!

I know that your firmware is for Wemos D1 mini. I also have the Wemos D1, but only one and it is already in use for my temperature sensors.
Are you planing on a release of the decoding part as a library?

Sure, I have no TX attached. ebusd also runs in readonly mode. But ebusd still wants to write to the bus and gives many errors about that.

I use a 5V@700mA power supply, that should be enough. I measured 5V@50mA power usage on my ESP8266 with your firmware.

Just tried it with my MacBook, and I got a few exceptions directly after reset:

```
Welcome to eBUS adapter 2.0, build 20180923
ebusd device string: xxx.xxx.xxx.xxx:9999
Press return within 5 seconds to change the configuration...
Entering eBUS mode.

Exception (0):
epc1=0x40211f14 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont
sp: 3ffffa00 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffba0:  400043e9 00000030 00000016 ffffffff
3ffffbb0:  400044ab 3fffc718 3ffffca0 08000000
3ffffbc0:  60000200 08000000 00000000 00000000

3ffffbd0:  0000ffff 00042035 00002035 003fe000
3ffffbe0:  40249fd0 0000049c 003fe000 00000030
3ffffbf0:  feefeffe feefeffe feefeffe feefeffe
3ffffc00:  5ffbbc23 feefeffe feefeffe feefeffe
3ffffc10:  00000000 00000000 0000001f 00000020
```
Probably caused my incompatible board - whatever.

So sorry for the noise, I will write my own decoder for the eBUS.

Thanks anyway.

closed 2018-09-30 by Xeno22