john30/ebusd-esp #13: Unexplained changes to Config

Joe2019, 2019-01-23:
Hi I am having problems with the EBUS adaptor 2.0(WMOS) Build 20181125.
I have it set to a fixed IP address  example 192.168.1.17 - and set the corresponding mask and gateway.  . Everything seems fine and I can access it ok with the IP address and connect to it ok.. 
After a few hours I rechecked it and its IP address has changed to a DCHP allocated  address.  
The web screen still shows the fixed IP settings i.e. 192.168.1.17 but its actual address I have to use to access the WEB interface is my next free DCHP address (192.168.1.102).  Changing the settings on the WEB screen or rebooting don’t change it back to the Fixed address.  The only way I can restore the Fixed IP address it to use the Firmware update option and reload the Firmware.. Then the WMOS reboots to the fixed IP address without any changes having to be made to the configuration screen.
Do any of the unused inputs on the WMOS  trigger anything - I have tied the RESET input to 3v3  in case this was the problem but other unused inputs are left floating.
I have checked for any power stability problems but this does not seem to be the problem.
Any suggestions welcome , alternatively could you let me see the source code so I can debug this myself..

john30, 2019-01-27:
I didn't see such a behaviour yet. I guess your wemos is damaged. no input pins are used for triggering anything.

john30, 2019-04-06:
can you check again with latest bin?

john30, 2019-11-10:
clsoed due to no response

closed 2019-11-10 by john30