john30/ebusd-esp #15: ebusd-esp does not reconnect to wifi

SebastianCornea, 2019-02-24:
HI John, 

I noticed random crashes of ebusd-esp. So far one that I can reproduce is when I restart the wifi access point or wifi is down for any reason and then back again, ebusd-esp will not reconnect to wifi or at least it's web config page is no longer available. Could be related to https://github.com/esp8266/Arduino/issues/4445 . 
I'm using the latest build 20181125.

SebastianCornea, 2019-02-25:
Update: it actually reconnects to wifi just that it uses DHCP instead of preset IP address so it will listen on a different IP. If I reboot it from the web interface it will connect again correctly using the preset IP.

john30, 2019-02-28:
do you have a power supply with enough amps? i know that Wemos is pretty unstable when the power supply is not good enough

SebastianCornea, 2019-02-28:
Yes, more than it actually uses. For now I fixed the issue by setting a permanent IP address given in DHCP for the wemos MAC. 
The behavior is like it is forgetting all the settings as I noticed that the host name also resets to default. Could be a power fluctuation.

john30, 2019-04-06:
not confirmed yet, thus reopening

john30, 2019-04-06:
can you check with latest bin?

SebastianCornea, 2019-04-07:
updated the firmware and disabled the DHCP fix. I couldn't reproduce the issue yet and I will keep it under observation for a while.

john30, 2019-04-07:
great. I've just updated the binary again using the latest SoftwareSerial release which might also affect the behaviour in case other connectors than RX/TX are being used

SebastianCornea, 2019-04-11:
So far the new version seems stable, no more wifi reconnect issues.

closed 2019-04-11 by SebastianCornea

john30, 2019-04-12:
good to know

mozzarelka, 2019-04-22:
@john30, Thanks for update, so far no more connection issues. Less communication errors during normal work also.

commit 7f1af899 (2019-04-06): added RSSI, updated library and SDK (might fix #15 and #13), added analog PIN (fixes #12)