john30/ebusd-esp #16: No fix IP and no serial connection

find-a-good-nickname, 2019-08-05:
Dear John30,

I'm a newbie on ebusd, and tried to flash your mod'd firmware (20190407) to a Wemos D1 Mini. 
I accessed the flashed fw with a cellphone and configured the WiFi to the RasPi's WiFi. After a restart Wemos connected to the RasPi correctly, but when I tried to give a fix IP address, save it and reboot, the Wemos stayed on the DHCP'd IP, and not the fix. I tried to reflash and set the fix IP again and again, but the firmware can't save the IP. (There was ESPEasy and Tasmota on the Wemos before, and those were working flawlessly with a fix IP.)

Ok, I tried then to connect Wemos on USB with Putty, but it can't: the Putty window is empty with the COM port (I flashed Wemos through this port, so the virtual COM9 is WORKING) and with 115200 bps.

What can I do to give the fix IP to the Wemos/eBUS adapter 2.0?

Thanks for any ideas!

john30, 2019-09-08:
there were issues with DHCP as reported in #15, but those were fixed in the latest release.
if you configure for a fix IP you have to enter the gateway IP as well. Otherwise it might just fall back to DHCP.
If you connect via Putty, you have to press the reset button after the putty window is open. then a welcome message with intructions should appear.
btw: what device are you using exactly?

find-a-good-nickname, 2019-09-09:
It is the latest release (20190407) and a Wemos D1 Mini.

Ah, thanks, I pressed the reset button when in Putty, and it welcomes to eBUS adapter 2.0 build 20190407, and shows IP (not fix) and WiFi signal. But when the "press return within 5 seconds to change the configuration" message appears, I press the Enter, the Wemos still writes "Entering eBUS mode" every time and nothing else. There is no menu or something.
[image]

And again: I wrote the gateway IP in the "gateway", but after a reboot the Wemos/eBUSD can't find and connect the WiFi. If I write the netmask (255.255.255.0) to the "gateway" it works well, connects to the WiFi in no time.  Look at the picture in my first message.
It seems that the "gateway" rubric is really the "netmask", and the real "gateway" is not visible at all.

john30, 2019-09-17:
you probably have to adjust the putty settings wrt line separator.
for the gateway issue I can't verify your findings, it works perfectly well for me in all kinds of combinations

john30, 2019-11-10:
please check again with the latest binary

john30, 2019-12-08:
the DHCP issue might be solved with the latest binary from today. Please check again

closed 2020-04-12 by john30