john30/ebusd-esp #17: Support 1-wire

andig, 2019-08-20:
I'm using the 2.1 adapter with attached WEMOS. As my heating does not supply VL temps for HK1 and HK2 I'd like to add those. 1-wire sensors with precision reading seem to be the obvious choice. 

In https://github.com/john30/ebusd-esp/commit/7f1af8990cabd20ebc567390f1df15a6d98084e8, analog input was added. Would it be possible to consider adding 1-wire as well, at least for testing purposes to see if it can still keep the ebus timing?

andig, 2019-09-06:
ping @john30 any chance giving this a try? I'd be happy to attempt this under NDA if that would help in any way spread the work.

john30, 2019-09-08:
I'm working on that. what class of 1W sensor do you want to use? Is it a DS1820?

andig, 2019-09-08:
Yes, I have a bunch of cheap ones like these: https://www.ebay.de/itm/5-10-Pcs-DS18B20-Wasserdicht-Thermometer-Temperaturfühler-Temperatursensor-Kabel/392177950949?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

john30, 2019-09-17:
I made some progress and was able to read a DS1820 temperature. The only thing is the complication of the number of sensors to be attached since every sensor has a long unique ID and you'd basically have to have the possiblity to configure each sensor in ebusd-esp to get like a fixed index or name or something. this makes the configuration quite complicated. so I'd rather do a scan for sensors after boot, calculate a checksum over the ID, sort the entries and that's it.

andig, 2019-09-17:
Sounds good. At least for me the attached sensors would be stable and I can adjust to the checksum. Given the temperatures it should also be easy to figure out which sensor is which. Please let me know when there's anything to test ;)

andig, 2019-09-24:
ping @john30 would be great to get this running before the heating period starts again ;)

john30, 2019-09-29:
fixed with 350b762

andig, 2019-09-29:
Do I have any chance to fetch this via "Check & Update"?

andig, 2019-09-29:
I have the new FW running now- but how would I connect the 1-wires? I.e. which pins should they attach to?

john30, 2019-09-29:
you need to configure a port for being used for sensors first. pull-up is needed as required by the corresponding hardware. I've tested it successfully with a pull-up for a Dallas temp sensor found somewhere in the web

andig, 2019-10-06:
Update: still working on the test setup. As I need to put terminals for the sensors into the enclosure now I'll rewire everything a little less space-consuming than I did before with headers mounted to the WeMos. Parts are ordered and I hope to get it done next weekend.

andig, 2019-10-06:
I have some problems testing this on an esp8266-201:

- esp not connecting wifi (claims ip is 255.255.255.255 over serial)
- cannot save initial pin config across reboot
- cannot find sensors (have these powered with 3.3V and 4.7k pull-up)
- saw one wdt reset on serial connecting wifi

To be sure these aren't fake I'll retry once the ordered Wemos arrives- hopefully next weekend.

john30, 2019-10-07:
don't know if this is due to the esp model you have. it works fine with my wemos 3.0

andig, 2019-10-07:
I‘ll retry with brand wemos. I‘ll also check if sensors work better when fed by 5v (2.5m wires attached) as it seems that the esp GPIOs are 5v tolerant.

andig, 2019-10-12:
Could you upload a preview build of the "lite" binary? Got the new D1 Mini Lites now, the old lite binary is running fine but the preview one just outputs garbage. 

Also, I have some problem using the config screen in the old image- seems it is entirely missing any kind of linebreaks regardless of terminal emulation I chose:

                                                                                                      Welcome to eBUS adapter 2.0, build 20190407
                                                                                                                                                 Configured as WIFI access point EBUS without password.
                                                         For configuration with web browser, connect to this WIFI and open http://192.168.4.1/
                                                                                                                                              Entering configuration mode.
                            Chip ID: 00CFB492
                                             Hostname: ebus-cfb492
    
                                                                  Configuration (new):
                                                                                       1. WIFI SSID: EBUS
                                                                                                          2. WIFI secret: 
                                                                                                                           3. WIFI IP address: DHCP (not connected)
                      4. WIFI hostname: ebus-cfb492
                                                    5. eBUS RX+TX PINs: direct RX+TX (GPIO3+1)
                                                                                               6. ebusd connection: TCP on port 9999
                                                                                                                                     7. HTTP TCP port: 80
            8. LED PINs: RX:disabled, TX:disabled
                                                  9. Initial PINs: D4:H
    
                                                                        p. Set current PINs: D4:H
                                                                                                  t. Toggle current output PIN
                                                                                                                               c. Connect WIFI
                                                                                                                                               e. Dump EEPROM content
                        f. Load factory settings
                                                 F. Factory reset (i.e. erase EEPROM)
                                                                                      o. OTA enabled: waiting
                                                                                                              r. Reboot (without saving)
                                                                                                                                         0. Save configuration and reboot
    
                           Enter your choice:

andig, 2019-10-12:
To be more precise regarding the line break: I don't think this is a terminal problem as the normal WDT/Reboot messages have line breaks:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld
```

john30, 2019-10-12:
simply activate "Implicit CR in every LF" and "Implicit LF in every CR" in Putty config, that's it.

new preview binaries are here (zipped):
[ebus-v2_d1mini-preview.zip](https://ebusd.eu/preview/ebus-v2_d1mini-preview.zip)
[ebus-v2_d1mini_lite-preview.zip](https://ebusd.eu/preview/ebus-v2_d1mini_lite-preview.zip)

andig, 2019-10-12:
> simply activate "Implicit CR in every LF" and "Implicit LF in every CR" in Putty config, that's it.

I'm on Mac using `serial`. Neither CRLF, CR nor LF settings work.

andig, 2019-10-12:
Using the new mini lite image- same as with the release mini lite image- I'm unable to connect to wifi (can open separate issue):

```
   Welcome to eBUS adapter 2.0, build 20191012
                                              ebusd device string: 255.255.255.255:9999
                                                                                       WiFi signal: 100% (-44dBm)
                                                                                                                 Press return within 5 seconds to change the configuration...
```

I do see it in the fritzbox config and it has an IP (or had one for a moment) but it doesn't answer to ping or http. I can't connect from serial menu either: it always shows `DHCP (connected: 255.255.255.255)` as IP.

andig, 2019-10-12:
>> simply activate "Implicit CR in every LF" and "Implicit LF in every CR" in Putty config, that's it.

> I'm on Mac using serial. Neither CRLF, CR nor LF settings work.

nvm. serial has the option, too:

[image]

andig, 2019-10-12:
Not only DHCP, also fixed IP is not working although it says "connected". There is no IP conflict on the network.:

```
Enter your choice: c
Connecting WIFI...
WIFI connected.

ebusd device string: 255.255.255.255:9999
WiFi signal: 100% (-47dBm)
Configuration (changed):
 1. WIFI SSID: XXXXXXXX
 2. WIFI secret: ****
 3. WIFI IP address: 192.168.0.87/24 (connected), gateway: 192.168.0.1
```

Using "connect" inside the menu does not fix this. IP remains at 255... in DHCP mode or at fixed IP if fixed. Device remains unaccessible.

Then... when I finally "save and rebooted" with fixed IP set, device came up. Changing back to DHCP afterwards (save & reboot) continue to work. It looks as if there is a problem with the initial DHCP network connection and maybe with the connect menu option.

andig, 2019-10-12:
OK. Now that I got WIFI working I've moved to sensor. I have one attached locally:

```
Sensors:
D1: 2817B650040000: temperature=127.50
1 sensors found
```

Now temp is at a wrong but stable value when sensor (on breadboard) is powered from 3.3V. 

When sensor has 5V connected (Wemos should be 5V tolerant), sensor is not readable. 4.7k pullup is connected.

andig, 2019-10-13:
> Update remaining thing that might be nice would be increased readout precision. Not sure it that would help with the offset issue but seems worth testing.

After reading some more I guess isn't. We already have 11bits (0.065°C) and the sensors precisio is apparently 0.5°C.

andig, 2019-10-18:
@john30 I have since moved my setup from breadboard to actual boiler. Downturn: sensors are no longer detected. I've checked the wiring without ripping everything apart again and it still looks ok.

Is there any chance that the new sensors support might not work when ebus adapter is actually connected? If not I'll disassemble and retry...

john30, 2019-10-19:
no, I've tested it. The sensors are only queried from ebusd-esp when requested

andig, 2019-11-03:
> Is there any chance that the new sensors support might not work when ebus adapter is actually connected? If not I'll disassemble and retry...

I've verified that the sensors are not detected/ not visible in the UI if RX/TX of an actual ebus adapter are attached to the WeMOS.

john30, 2019-11-07:
it works fine on mine with sensors connected on D7. will try to verify your setup with sensors on D7

john30, 2019-11-10:
it also works fine for me when setting D1 to be used for sensors. Anyway, I've updated the binaries in c8c2916 which also includes an updated SoftwareSerial library, so please try again with those.

john30, 2019-12-08:
the DHCP issue might be solved now. It seems a hard set to not using an IP was potentially missing

closed 2021-10-30 by john30