john30/ebusd-esp #18: Nodemcu unreachable after entering eBUS mode

mbbelgium, 2019-11-03:
I've been testing 2 days so far and I don't get the ebus adapter working. I flashed the ebus-v2_nodemcu1.bin using the nodemcu firmware programmer. It starts good, and can enter the setup. After the reboot, the nodemcu makes a coonection to the WIFI, but after 5 seconds (when entering eBUS mode) no communication possible. I can't ping the device, no connection is made, and nothing in the serial monitor eather... Someone else had this problem? @john30 , any idea what can be wrong here? Is the nodemcu waiting for something in eBUS mode before starting up a network connection? I can't find the sourcefile to see where it can go wrong.
Thanks in advance for helping me out!

mbbelgium, 2019-11-09:
I finally ordered a Wemos d1 mini and it's working. So nodemcu is not working. Just for information

john30, 2019-11-10:
please post your configuration of both the ebusd-esp and ebusd together with the info how you have wired the adapter to the Nodemcu board. please also try with the updated binary

mbbelgium, 2019-11-10:
Hi John, what I meant is that the nodemcu doesn't get an IP once entering EBUS mode. But with the wemos, no problem. I will try the updated binary on a nodemcu, just in case someone has the same problem. I will keep you informed!

mbbelgium, 2019-11-11:
Hi John, I tested your new binarie on the nodemcu, and now it works. Thank you for your work.

closed 2019-11-11 by mbbelgium

john30, 2019-11-11:
great, thanks for testing again!