john30/ebusd-esp #21: Not possible to change TCP/IP...

darekhome, 2019-12-10:
Something is wrong? Or I'm doing something wrong?
I'm not able to change the TCP/IP - even I enter from 'putty' or from 'browser'
the address is not changing.
I want to change to '135' but the board is always appearing under '237'.

P.S. The board used: WEMOS D1 mini; today I specially purchased it as I had problems with mini PRO.

[image]

john30, 2019-12-13:
can you please try again with the latest binary as of bb20455 ?

darekhome, 2019-12-13:
O.K. Now looks v. good. Checked.
One more question - why in the terminal the 'print-out' is not adjusted?
[image]

john30, 2019-12-14:
great.
you need to adjust the putty settings to have implicit cr in every lf and implicit lf in every cr

closed 2019-12-14 by john30