john30/ebusd-esp #22: Sensor detected when Wemos connected by USB, but this state is not appearing in HTML GUI...

darekhome, 2019-12-14:
Is that some bug?
`Enter your choice: s
Sensors:
D4: 288D3C94971203: temperature=23.69
1 sensors found
`

[image]

darekhome, 2019-12-14:
Some update - when 'set-up' sensor with GUI and reboot - no sensor seen...
`{"sensors": []}`

[image]

john30, 2019-12-15:
the "s" is just for checking all ports for any connected sensor. you'd still have to set up the desired port as sensor port using "p" or "set initial" to sensor in HTML UI and then save and reboot to make it permanent

darekhome, 2019-12-15:
O.K, but when we focus on HTML... I have initial situation - sensor connected to 'D5'.
[image]

Set-up screen looks like above.

Then I click on at 'D5' 'Set Initial' 'Sensor' and then the 'Current' also is changing to 'Sensor'.
So screen looks like below:

[image]

Then I do 'Check & Update' and then 'Save & Reset'.

[image]

After reboot I see:

[image]

But still no sensor at 'HTML'...

`{"sensors": []}`

john30, 2019-12-15:
in your previous post the serial reported the sensors on D4

darekhome, 2019-12-15:
Yes, I tried different pins, but result the same. Now it is at 'D5' - see proof.
And latest post - tested with D5.

[image]

john30, 2019-12-16:
I'll try to reproduce at the weekend. Support for sensors is experimental anyway as stated in the readme

darekhome, 2019-12-16:
Yes, I know... I'm aware of that. Thx...

john30, 2019-12-25:
did you power the 1wire sensor appropriately?

john30, 2019-12-25:
it works fine for me, also on D5. If you change the PIN in the HTML UI, the sensors appear immediately (and the sensors tab) if they were scanned successfully

darekhome, 2019-12-26:
O.K. I'll check again. Also the supply of the sensor.

closed 2020-04-12 by john30