john30/ebusd-esp #23: Not saved configuration in 'PRO' version...

darekhome, 2019-12-15:
I wanted to test 'PRO' version, but it is some problem with saving the configuration.
When I set-up with USB, all goes O.K. I.e. wi-fi is connecting.
[image]

But when opening the page, doing some additional changes, etc. and then saving...

[image]

After reboot it is not possible to connect with the IP defined.

But when (again) connecting via USB - the screen is initial, looks like no data saved...

[image]

=================
Additional 'small-bug' on the initial screen 'host-name' is not updated, even if defined by user...
It is still default 'ebus-ct...'.
But on next screen it is O.K., given name by user 'ebus'.

[image]

[image]

john30, 2019-12-25:
both works fine for me, but I can't test on a pro as I don't own one

blue-beard, 2020-01-29:
I confirm on the ebus-v2_d1mini_pro.bin firmware the wifi configuration is not saved, with other firmware the module starts at all for me. tried through serial and web interface
d1 mini from this order https://forum.fhem.de/index.php/topic,93190.0.html

john30, 2020-01-30:
that should not be a "pro" version

blue-beard, 2020-01-30:
> это не должно быть "про" версия

I understand, on the ESP8266MOD chip, but it starts only with pro firmware, several times yesterday I tried all the firmware in a circle
[image]

john30, 2020-02-02:
please post the output on serial after you flashed with the right firmware and rebooted

blue-beard, 2020-02-03:
I apologize for the concern ebus-v2_d1mini.bin has earned and saves everything

john30, 2020-02-05:
isse can be closed then I assume?

closed 2020-03-10 by john30