john30/ebusd-esp #24: No reception with GPIO3/RX (ebus-v2_d1mini.bin)

B-G-T, 2020-01-19:
If the output of the self-made ebus adapter V1.6 is connected to RX, the signal does not reach low level (confirmed by oscilloscope) which means the direct RX mode does not work.

Everything works fine if D1 is used for reception (in combination with the setting SW D1).

john30, 2020-02-02:
not sure if this will work anyway as V1.6 of the interface surely has different logical power levels

B-G-T, 2020-02-02:
Frankly speaking I wonder if the RX line is maybe misconfigured as output (always high) instead of input?
You don't plan to make the ESP program open source, do you?

john30, 2020-04-11:
surely not. as said, I'm not sure if this will work with the 1.6 hardware

woutput, 2020-05-17:
> You don't plan to make the ESP program open source, do you?

@john30 could you explain why please? Preferably in README.md? Thank you.