john30/ebusd-esp #25: eBUS Adapter 2.2: eBUS signal: no signal

keiser1080, 2020-02-03:
hi friend,

I need some help.
I just trie to connect the ebus adapter2.2 today (buyed batch december 2018).
But no success.
i have installed the last firmware ebusd-esp Build: 20191229

I removed the extention board.
connected my vaillant turbotech pro to the ebus terminal on base board.
Powered the wemos using the usb interface.

on the ebus-esp i see:
ebusd connected: yes (inactive)
eBUS signal: no signal

Is there any polarity for the ebus on the base board.
For info my ebus thermostat work perfectly, i have tried to install both in parallel and separately same issue.
here the command used to launch the ebus damon ./src/ebusd/ebusd -d 192.168.2.225:9999 -l /var/log/ebusd.log --scanconfig --latency=20000 --address=01

Could you help?

keiser1080, 2020-02-04:
The problem is solve.
The ebus header was broken, and the yellow ligth still shine so it was diffifukt to found the issue.

closed 2020-02-04 by keiser1080

jacekk233, 2022-10-10:
Hello,
I built the same adapter, and have the same problem.
Could you describe me shortly what was in your case ?

jacekk233, 2022-12-08:
Hello
   
  
  
   
  
  
   Thanks for your answer. 
  
  
   I've mounted adapter on my own boardand in between times i solved the problem. 
  
  
   Few time ago I've bought some Wemos- differrent than usual.
  
  
   I spoted that ESP was mounted directly to the main board, and without one transistor there, which just made shorts on RX line.
   
  
  
   After change device works properly
   
  
  
   
  
  
   Thanks

   
   
    
   
   
    
   
   
    

    
   
   
    hi, i just see your message the first time it was the blue screw terminal in the write in this picture:
   
   
    [](https://adapter.ebusd.eu/v2/img/base-top-v20.jpg)
   
   
    The second time i have found some bad solder contact on a chip.
   
   
    
   
   
   
   
   
   
    https://github.com/john30/ebusd-esp/issues/25#issuecomment-1341492124