john30/ebusd-esp #27: link to instructions for connecting to ebus?

perfectom, 2020-02-22:
Hi, this firmware looks great! Please is it possible to include links to instructions how to wire NodeMCU (and the other module types) ?  Just to complete the picture..   I have the firmware loaded on my nodemcu but I don't know how to wire it to my heater.  I understand it's possible to make a simple read-only ebus interface with these instructions:https://ebus-wiki.org/lib/exe/fetch.php/ebus/arduinoebus.pdf  Is it possible to use this with it?

Thank you 
Tom

john30, 2020-03-10:
you can basically link it in several ways as the pins to be used are configurable to a certain extent. usually the RX+TX pins are used. wrt the interface you mentioned, I don't know if this works. just try.

woutput, 2020-05-17:
I agree with @perfectom I also think this would make this (for me potentially) very useful repository even more useful.
@john30 would you be open to pull requests that would add this information to README.md for example?

eiten, 2020-06-15:
Hello John30,

so https://easyeda.com/slavikb/EBus_Controller-26362750260a4f6baa63b59b5b35698d should work then as an interface, if we leave the LEDs away?

Regards, Edi

john30, 2020-09-20:
I guess this should work, but you need to try

vlad-323, 2020-12-15:
> Hello John30,
> 
> so https://easyeda.com/slavikb/EBus_Controller-26362750260a4f6baa63b59b5b35698d should work then as an interface, if we leave the LEDs away?
> 
> Regards, Edi

Hello. Maybe this is not actual, but this schematic is working, I repeated this adapter.

john30, 2021-02-20:
> Hello. Maybe this is not actual, but this schematic is working, I repeated this adapter.

may be, but it is a very old design that consumes too much power from the eBUS. We have a [brand new design now](https://adapter.ebusd.eu/index.en) doing that a lot better

closed 2021-02-20 by john30