john30/ebusd-esp #29: weird ebus messages

copystring, 2020-04-29:
I have got this board: https://ebus.github.io/adapter/raspberrypi
When connect to my RPi it works perfectly fine: 
[image]

But I do not want to run this via the RPi anymore. So I hooked it up to a wemos d1 mini.
5V to 5V
3.3V to 3.3V
TX to RX
RX to TX
(also tried TX to TX and RX to RX)
GND to GND

Now it does not work when connected to the d1 mini with https://github.com/john30/ebusd-esp as firmware.

[image]

This is the config:

[image]

Any ideas what could be wrong? I check wiring several times. Looks good to me.

john30, 2020-05-10:
what does the second putty window show, i.e. how did you start it?

copystring, 2020-05-10:
The second windows is connected via TCP on port 9999.

john30, 2020-05-10:
ok, so what do you expect to see there? it now shows the raw binary traffic, which is not what you usually want to do. instead you'd point your ebusd to the wemos by using the start parameter "--device tcp:ebus:9999" (since "ebus" seems to be the hostname of the wemos) and then the ebusd log file should show more or the same as before.

copystring, 2020-05-10:
Hm okay. I added tcp: in front so it looks like this:
`EBUSD_OPTS="-d tcp:ebus_wemos:9999 --configpath=/etc/ebusd/wolf/ -p 8888 -l /var/log/ebusd.log --scanconfig --httpport=8080 --mqttport=1883 --mqttjson --mqtthost=192.168.1.221 --mqttclientid=ebusd --mqtttopic=ebusd/%circuit/%name --accesslevel=* --enablehex"`

hostname of wemos is ebus_wemos

Is there not a way to see the ebus messages instead of the tcp messages like this?

closed 2020-05-10 by copystring