john30/ebusd-esp #30: Is lolin32 really correct for ESP-WROOM-32?

kawadik55, 2020-10-17:
Hi! I have had to download and configure lolin32 in the ESP-WROOM-32. This module was assembled with the eBUS adapter on one board. Connected the module through TCP to ebusd (RPI4). On the eBUS there is a Vaillant boiler only, which is sent AUTO-SYN. I connected a UART sniffer on the Rx WROOM, where both Tx and Rx (zero modem on the eBUS adapter) can be seen. ebusd started full-scan:
EBUSD_OPTS="--scanconfig=full -d 192.168.1.101:9999 --answer --localhost -p 8888 --httpport=8889 --enablehex --logfile=/home/pi/Desktop/shara/ebusd.log --loglevel=debug"
and sees a mismatch between the sent commands from ebasd and the transmitted bytes from the WROOM in eBUS. Is there a precedent for WROOM to work correctly with lolin32 firmware?

kawadik55, 2020-10-17:
log:
2020-10-17 16:14:01.246 [main notice] starting initial full scan
2020-10-17 16:14:01.246 [bus info] scan 02 cmd: 3102070400
2020-10-17 16:14:01.280 [bus debug] arbitration delay 128 micros
2020-10-17 16:14:01.280 [bus info] arbitration delay 128 - 128 micros
2020-10-17 16:14:01.280 [bus debug] switching from ready to send command
2020-10-17 16:14:01.310 [bus debug] notify request: ERR: read timeout
2020-10-17 16:14:01.310 [bus info] scan 04 cmd: 3104070400

hex:
AA AA AA AA AA AA AA AA   AA AA AA AA AA AA AA AA
31 AA AA 02 31 31 AA 31   AA AA 06 31 31 AA 31 AA   and so on.

kawadik55, 2020-10-19:
Using the logger, I see that Tx sometimes climbs onto AUTO-SYN.

john30, 2021-02-20:
start ebusd with "--lograwdata=bytes" and provide a minute of logging please

kawadik55, 2021-02-20:
For some reason, the log file is not attached.

pi@iob:~ $ sudo ebusd --scanconfig=full -f -d 192.168.3.101:8880 --lograwdata=bytes --answer --accesslevel="*" --latency=100000 --loglevel=debug > ebusd.log
2021-02-20 19:43:44.983 [main notice] ebusd 21.2.v21.2 started with full scan on device 192.168.3.101:8880
2021-02-20 19:43:44.984 [main info] loading configuration files from http://ebusd.eu/config/
2021-02-20 19:43:45.149 [main info] reading templates /
2021-02-20 19:43:45.308 [main info] read templates in /
2021-02-20 19:43:45.308 [main info] reading file memory.csv
2021-02-20 19:43:45.520 [main info] successfully read file memory.csv
2021-02-20 19:43:45.520 [main info] reading file broadcast.csv
2021-02-20 19:43:45.669 [main info] successfully read file broadcast.csv
2021-02-20 19:43:45.669 [main info] read config files
2021-02-20 19:43:45.992 [bus notice] bus started with own address 31/36 in answer mode
2021-02-20 19:43:45.993 [main info] registering data handlers
2021-02-20 19:43:45.993 [main info] registered data handlers
2021-02-20 19:43:46.045 [bus notice] <aa
2021-02-20 19:43:46.045 [bus debug] ERR: SYN received during no signal, switching to ready
2021-02-20 19:43:46.045 [bus notice] signal acquired
2021-02-20 19:43:46.088 [bus notice] <aa
2021-02-20 19:43:46.136 [bus notice] <aa
2021-02-20 19:43:46.174 [bus notice] <aa
2021-02-20 19:43:46.217 [bus notice] <aa
2021-02-20 19:43:46.268 [bus notice] <aa
2021-02-20 19:43:46.308 [bus notice] <aa
2021-02-20 19:43:46.353 [bus notice] <aa
2021-02-20 19:43:46.395 [bus notice] <aa
2021-02-20 19:43:46.439 [bus notice] <aa
2021-02-20 19:43:46.477 [bus notice] <aa
2021-02-20 19:43:46.521 [bus notice] <aa
2021-02-20 19:43:46.566 [bus notice] <aa
2021-02-20 19:43:46.611 [bus notice] <aa
2021-02-20 19:43:46.650 [bus notice] <aa
2021-02-20 19:43:46.694 [bus notice] <aa
2021-02-20 19:43:46.740 [bus notice] <aa
2021-02-20 19:43:46.780 [bus notice] <aa
2021-02-20 19:43:46.826 [bus notice] <aa
2021-02-20 19:43:46.867 [bus notice] <aa
2021-02-20 19:43:46.911 [bus notice] <aa
2021-02-20 19:43:46.954 [bus notice] <aa
2021-02-20 19:43:46.998 [bus notice] <aa
2021-02-20 19:43:47.048 [bus notice] <aa
2021-02-20 19:43:47.087 [bus notice] <aa
2021-02-20 19:43:47.127 [bus notice] <aa
2021-02-20 19:43:47.170 [bus notice] <aa
2021-02-20 19:43:47.219 [bus notice] <aa
2021-02-20 19:43:47.266 [bus notice] <aa
2021-02-20 19:43:47.300 [bus notice] <aa
2021-02-20 19:43:47.344 [bus notice] <aa
2021-02-20 19:43:47.390 [bus notice] <aa
2021-02-20 19:43:47.431 [bus notice] <aa
2021-02-20 19:43:47.474 [bus notice] <aa
2021-02-20 19:43:47.516 [bus notice] <aa
2021-02-20 19:43:47.560 [bus notice] <aa
2021-02-20 19:43:47.604 [bus notice] <aa
2021-02-20 19:43:47.650 [bus notice] <aa
2021-02-20 19:43:47.703 [bus notice] <aa
2021-02-20 19:43:47.734 [bus notice] <aa
2021-02-20 19:43:47.777 [bus notice] <aa
2021-02-20 19:43:47.829 [bus notice] <aa
2021-02-20 19:43:47.871 [bus notice] <aa
2021-02-20 19:43:47.907 [bus notice] <aa
2021-02-20 19:43:47.952 [bus notice] <aa
2021-02-20 19:43:47.994 [bus notice] <aa
2021-02-20 19:43:48.037 [bus notice] <aa
2021-02-20 19:43:48.081 [bus notice] <aa
2021-02-20 19:43:48.129 [bus notice] <aa
2021-02-20 19:43:48.167 [bus notice] <aa
2021-02-20 19:43:48.218 [bus notice] <aa
2021-02-20 19:43:48.262 [bus notice] <aa
2021-02-20 19:43:48.301 [bus notice] <aa
[... 1712 lines cut ...]
2021-02-20 19:44:54.947 [bus notice] <aa
2021-02-20 19:44:54.947 [bus notice] <aa
2021-02-20 19:44:54.947 [bus notice] <aa
2021-02-20 19:44:54.947 [bus notice] <aa
2021-02-20 19:44:54.947 [bus notice] <aa
2021-02-20 19:44:54.947 [bus notice] <aa
2021-02-20 19:44:54.947 [bus notice] <aa
2021-02-20 19:44:55.996 [main debug] performing regular tasks
2021-02-20 19:44:56.035 [bus debug] ERR: read timeout during skip, switching to no signal
2021-02-20 19:44:56.035 [bus error] signal lost
2021-02-20 19:44:57.970 [bus notice] <aa
2021-02-20 19:44:57.970 [bus debug] ERR: SYN received during no signal, switching to ready
2021-02-20 19:44:57.970 [bus notice] signal acquired
2021-02-20 19:44:57.970 [bus notice] <aa
2021-02-20 19:44:58.138 [bus notice] <aa
2021-02-20 19:44:58.138 [bus notice] <aa
2021-02-20 19:44:58.138 [bus notice] <aa
2021-02-20 19:44:58.588 [bus notice] <aa
2021-02-20 19:44:58.589 [bus notice] <aa
2021-02-20 19:45:00.039 [bus debug] ERR: read timeout during skip, switching to no signal
2021-02-20 19:45:00.039 [bus error] signal lost
2021-02-20 19:45:05.996 [main debug] performing regular tasks
2021-02-20 19:45:06.276 [bus notice] device invalid
2021-02-20 19:45:10.340 [main notice] SIGINT received
2021-02-20 19:45:11.097 [main notice] ebusd stopped

john30, 2021-02-20:
it seems the interface is rarely able to write on the bus. this way it can't work reliably.
do you have the latest version of everything?

kawadik55, 2021-02-21:
Yes, all versions are up to date (adapter 3 and armhf). I see oscillograms, how the command is transmitted from, and how SYN comes in. Levels in both cases fall from ~ 20V to ~ 9V. In addition, I see in the logger these signals on Rx ESP32, and transmission and reception due to feedback on the ebus interface side. I also see timing inconsistency with the ebus protocol. If there was an opportunity, I would send logger reports and oscillograms, but this editor refuses to link files.

john30, 2021-02-21:
ah, then post the settings of ebusd-esp please

kawadik55, 2021-02-21:
pi@iob:~ $ sudo ebusd --scanconfig=full -f -d 192.168.3.101:8880 --lograwdata=bytes --answer --accesslevel="*" --latency=100000 --loglevel=debug > ebusd.log

kawadik55, 2021-02-23:
I think the SYNC frame synchronization needs to be done on the ESP32 side. Large TCP delays cause EBUS protocol failure.

john30, 2021-02-23:
that's exactly why we have the adapter 3 and a special esp firmware.
actually I was asking for your esp settings, i.e. the settings made on the esp device on the ebusd-esp firmware.
and according to ebusd settings you're not using the enhanced mode, which you should of course

kawadik55, 2021-02-23:
[image]
[image]
Was it necessary to do a "factory reset"? All settings remain from the previous version.
what does it mean "enhanced mode"? daemon?

kawadik55, 2021-02-23:
Additionally, I checked my ebus interface through ttyUSB0(usb2ttl) by disconnecting Tx-Rx from esp32. Ebusd v21.2 + ttyUSB0 works great.

kawadik55, 2021-02-25:
I did a "factory reset" and found enh: mode :)
[image]
But it didn't help much.

> pi@iob:~ $ ebusd --scanconfig=08 -f -d enh:192.168.3.101:9999 --answer --enablehex latency=100000 --lograwdata=bytes --loglevel=debug > ebus.log
2021-02-25 18:53:35.152 [main notice] ebusd 21.2.v21.2 started with single scan on enhanced device 192.168.3.101:9999
2021-02-25 18:53:35.153 [main info] loading configuration files from http://ebusd.eu/config/
2021-02-25 18:53:35.325 [main info] reading templates /
2021-02-25 18:53:35.573 [main info] read templates in /
2021-02-25 18:53:35.573 [main info] reading file memory.csv
2021-02-25 18:53:35.761 [main info] successfully read file memory.csv
2021-02-25 18:53:35.761 [main info] reading file broadcast.csv
2021-02-25 18:53:35.962 [main info] successfully read file broadcast.csv
2021-02-25 18:53:35.962 [main info] read config files
2021-02-25 18:53:36.287 [bus notice] bus started with own address 31/36 in answer mode
2021-02-25 18:53:36.287 [main info] registering data handlers
2021-02-25 18:53:36.287 [main info] registered data handlers
2021-02-25 18:53:36.292 [bus notice] device status: reset
2021-02-25 18:53:36.309 [bus notice] <aa
2021-02-25 18:53:36.309 [bus debug] ERR: SYN received during no signal, switching to ready
2021-02-25 18:53:36.309 [bus notice] signal acquired
2021-02-25 18:53:36.356 [bus notice] <aa
2021-02-25 18:53:36.402 [bus notice] <aa
.
2021-02-25 18:53:46.233 [bus notice] <aa
2021-02-25 18:53:46.275 [bus notice] <aa
2021-02-25 18:53:46.288 [main debug] performing regular tasks
2021-02-25 18:53:46.288 [main notice] starting initial scan for 08
2021-02-25 18:53:46.288 [bus info] scan 08 cmd: 3108070400
2021-02-25 18:53:46.316 [bus notice] <aa
2021-02-25 18:53:46.316 [bus debug] start request 31
2021-02-25 18:53:46.316 [bus debug] arbitration start with 31
2021-02-25 18:53:46.359 [bus notice] <aa
2021-02-25 18:53:46.369 [bus notice] >31
2021-02-25 18:53:46.369 [bus notice] <31
2021-02-25 18:53:46.369 [bus debug] arbitration won
2021-02-25 18:53:46.369 [bus debug] arbitration delay 12 micros
2021-02-25 18:53:46.369 [bus info] arbitration delay 12 - 12 micros
2021-02-25 18:53:46.369 [bus debug] switching from ready to send command
2021-02-25 18:53:46.369 [bus notice] >08
2021-02-25 18:53:46.417 [bus notice] <aa
2021-02-25 18:53:46.417 [bus debug] notify request: ERR: SYN received
2021-02-25 18:53:46.417 [bus debug] ERR: SYN received during send command, switching to ready
2021-02-25 18:53:46.418 [main error] initial scan failed: ERR: SYN received
2021-02-25 18:53:46.456 [bus notice] <aa
2021-02-25 18:53:46.478 [bus notice] <08
2021-02-25 18:53:46.522 [bus notice] <aa
2021-02-25 18:53:46.522 [bus debug] ERR: SYN received during receive command, switching to ready
2021-02-25 18:53:46.570 [bus notice] <aa
2021-02-25 18:53:46.612 [bus notice] <aa
2021-02-25 18:53:46.656 [bus notice] <aa
.
2021-02-25 18:54:12.995 [bus notice] <aa
2021-02-25 18:54:14.082 [bus debug] ERR: read timeout during skip, switching to no signal
2021-02-25 18:54:14.082 [bus error] signal lost
2021-02-25 18:54:14.315 [bus notice] <aa
2021-02-25 18:54:14.315 [bus debug] ERR: SYN received during no signal, switching to ready
2021-02-25 18:54:14.315 [bus notice] signal acquired
2021-02-25 18:54:14.319 [bus notice] <aa
.
2021-02-25 18:54:16.556 [bus notice] <aa
2021-02-25 18:54:16.556 [main notice] ebusd stopped

[image]

john30, 2021-04-11:
with adapter 3, you should put ebusd-esp into the "adapter 3" mode of course and adjust the adapter jumpers [as described here](https://adapter.ebusd.eu/index.en.html#wifi)

kawadik55, 2021-04-12:
Thank you, John! I use the not quite full-fledged "adapter 3," I have DIY according to this scheme:

[image]
So I think that the task "enhanced mode" is not divided between ESP and PIC in "adapter 3"?

The ESP configuration is configured as you said:

[image]

As a result, this is:

pi@linux:~ $ sudo ebusd --scanconfig -f -d enh:192.168.3.101:9999 --loglevel=debug
2021-04-12 16:07:33.505 [main notice] ebusd 21.2.v21.2 started with auto scan on enhanced device 192.168.3.101:9999
2021-04-12 16:07:33.505 [main info] loading configuration files from http://ebusd.eu/config/
2021-04-12 16:07:33.641 [main info] reading templates /
2021-04-12 16:07:33.778 [main info] read templates in /
2021-04-12 16:07:33.778 [main info] reading file memory.csv
2021-04-12 16:07:33.916 [main info] successfully read file memory.csv
2021-04-12 16:07:33.916 [main info] reading file broadcast.csv
2021-04-12 16:07:34.055 [main info] successfully read file broadcast.csv
2021-04-12 16:07:34.055 [main info] read config files
2021-04-12 16:07:34.164 [bus notice] bus started with own address 31/36
2021-04-12 16:07:34.164 [main info] registering data handlers
2021-04-12 16:07:34.164 [main info] registered data handlers
2021-04-12 16:07:34.173 [bus notice] device status: reset
2021-04-12 16:07:34.679 [bus error] **device status: unexpected available enhanced byte 2**
2021-04-12 16:07:34.682 [bus notice] signal acquired
2021-04-12 16:07:34.719 [bus error] **device status: unexpected enhanced command 0x0e**
2021-04-12 16:07:34.762 [bus error] device status: unexpected enhanced command 0x0e
...

Is there something wrong with enhanced mode?

HeikoGr, 2021-04-14:
Yes, there is something wrong - because the enhanced mode is realized in the PIC. I wouldn't call your adapter Version 3 according to the provided sheet. Maybe disabling enhanced mode helps.

kawadik55, 2021-04-15:
Ok! Thank you! Zur Rahmat!

closed 2021-06-06 by john30