john30/ebusd-esp #32: Add password security access to web configuration

vlad-323, 2020-12-15:
Hello. Thanks for your project! I want to ask - is it possible to add, like, basic authentication access to web configuration page? Because any user in local network can change parameters, I think this is security issue. P.S. Sorry for my English.

bobrippling, 2023-07-24:
I'd be interested in this too, and willing to do some development.

I was thinking of an extra configuration setting to create a password, and then have the ebus adapter require that via basic auth (admittedly still over HTTP, but without an existing connection it would be sufficiently safe for my use-case).

If this sounds ok to you all, I'll see what I can do.

john30, 2023-09-23:
not really a secure solution, but anyway part of the firmware now

closed 2023-09-23 by john30