john30/ebusd-esp #33: RFC: Select WiFi mode in configuration

mike-k68, 2020-12-30:
Hi guys. First of all many thanks for the great job.

There are connection issues with ESP8266 and some hi-performance routers, which I personnally experienced with Netgear R8000 and Asus RT-AX92U. The ESP8266 would hardly connect to the router (or not connect at all).
This can be solved easily by changing the WiFi mode of the ESP8266.
With my router I need to use mode G for a flawless usage.

So please here is my request:
Would you please add a new config item « WiFi mode » with values B,G and N and use it in the setup routine, depending on the config:
WiFi.setPhyMode(WIFI_PHY_MODE_11B);
or WiFi.setPhyMode(WIFI_PHY_MODE_11G);
or WiFi.setPhyMode(WIFI_PHY_MODE_11N);

With that I would be able to use ebusd-esp with my router.

Many thanks
Mike

netweaver1970, 2021-04-21:
Same issue here, my router (Netgear Nighthawk) doesn't seem to be liked by the ebus adapter v3 + WEmos D1 mini, the config is not saved as expected and the Wemos presents itself every time (after needing a reset) back with the default EBUS SSID.
Now, I also have a space in my home SSID. I'll have to search for a cheapo crappy old 2.4GHz router to see if I can at least progress somehow and if the space in my SSID is maybe an issue. Thanks

netweaver1970, 2021-04-21:
> Same issue here, my router (Netgear Nighthawk) doesn't seem to be liked by the ebus adapter v3 + WEmos D1 mini, the config is not saved as expected and the Wemos presents itself every time (after [...]

my issue was unrelated, It turned out the only to get in a working state again was doing a (re)flash of the binary.
Issue solved, wifi config is being saved.

commit 349b1b9c (2021-10-30): updated to 20211030 (fixes #48, fixes #33), formatting, hint to erase before flash
closed 2021-10-30 by john30