john30/ebusd-esp #35: reliable startup sequence?

goosst, 2021-02-20:
Hello,

thanks for the ebusd software, I'm using it for a year or two now on my vaillant heater. 

When my router reboots and new IP addresses get assigned, it's never able to connect to the heater immediately, until it randomly starts working again (but we're now already a few days later ...). Nothing changed in the hardware or its connections.
I've tried rebooting the devices, regular restarts with `sudo systemctl restart ebusd`, ... .
Is there some reliable startup procedure to get connection to the ebus?

I'm running the ebusd-esp software on a wemos with the v2 adapter, the ebusd software is running on a tinkerboard (this points to the IP address of the ebusd-esp).

Some more information below:

in /etc/default/ebusd I use
`EBUSD_OPTS="-d 192.168.0.193:9999 -l /var/log/ebusd.log --scanconfig --latency=50000 --address=01"`

The ebusd-esp adapter reports:
```
Build: 20191110
ebusd device string: 192.168.0.193:9999

Hostname: ebus-e387f5
Up time: 25445
Free heap: 30472
ebusd connected: no
eBUS signal: no signal
eBUS initial signal: yes 
```

some part of the ebusd.log:

```
2021-02-20 18:53:47.114 [main notice] SIGTERM received
2021-02-20 18:53:47.268 [main notice] ebusd 21.2.v21.2 started with auto scan on device 192.168.0.193:9999
2021-02-20 18:53:47.596 [bus notice] bus started with own address 01/06
2021-02-20 18:53:51.899 [main notice] ebusd stopped
2021-02-20 18:55:47.505 [main notice] ebusd 21.2.v21.2 started with auto scan on device 192.168.0.193:9999
2021-02-20 18:55:47.836 [bus notice] bus started with own address 01/06
2021-02-20 18:55:57.671 [main notice] update check: OK
2021-02-20 18:57:57.908 [main notice] update check: OK
2021-02-20 18:58:03.619 [main notice] SIGTERM received
2021-02-20 18:58:03.704 [main notice] ebusd 21.2.v21.2 started with auto scan on device 192.168.0.193:9999
2021-02-20 18:58:04.031 [bus notice] bus started with own address 01/06
2021-02-20 18:58:08.229 [main notice] ebusd stopped
2021-02-20 19:00:14.099 [main notice] update check: OK
2021-02-20 19:05:44.479 [main notice] ebusd 21.2.v21.2 started with auto scan on device 192.168.0.193:9999
2021-02-20 19:05:44.841 [bus notice] bus started with own address 01/06
2021-02-20 19:07:54.914 [main notice] update check: OK
2021-02-20 19:09:20.264 [main notice] SIGTERM received
2021-02-20 19:09:20.379 [main notice] ebusd 21.2.v21.2 started with auto scan on device 192.168.0.193:9999
2021-02-20 19:09:20.700 [bus notice] bus started with own address 01/06
2021-02-20 19:09:25.305 [main notice] ebusd stopped
2021-02-20 19:11:30.772 [main notice] update check: OK
```

ebusctl info
```
stijn@tinkerboard:/etc/default$ ebusctl info
version: ebusd 21.2.v21.2
update check: OK
signal: no signal
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd
```

john30, 2021-02-21:
is your wemos on DHCP? if so, then you'd have to use a name instead of the IP that gets resolved to the IP for the ebusd config

goosst, 2021-02-24:
thanks for the response, I have a router with very limited configuration options (or options in general).
But is there a known preference to startup the different components in the system (ebus-esp,ebusd, heater)? Or is my setup just flaky for some reason.

goosst, 2021-03-11:
I'm actually not understanding anymore how to interact with ebusd (not sure if it's linked to a latest update).

if I look with `tail -30 /var/log/ebusd.log`, I get the following messages, where it seems to be reading information and detecting the devices:
```
2021-03-11 19:50:47.411 [update notice] received read vr_70 SensorData QQ=10: -;-;-;-;45.94;27.69;c8 7d 00
2021-03-11 19:50:47.681 [update notice] received update-write bai SetMode QQ=10: auto;47.0;-;-;0;0;0;0;0;0
2021-03-11 19:50:47.938 [update notice] received read bai Status02 QQ=10: on;60;75.0;70;65.0
2021-03-11 19:50:48.174 [update notice] received update-read broadcast vdatetime QQ=10: 20:50:46;11.03.2021
2021-03-11 19:50:48.438 [update notice] received unknown MS cmd: 1008b5110100 / 080c031922040f0001
2021-03-11 19:50:48.697 [update notice] received write vr_70 SetActorState QQ=10: on;-;off;off;-;-;off
2021-03-11 19:50:48.943 [update notice] received write vr_70 Mc2FlowTempDesired QQ=10: on;28.0;on;5
2021-03-11 19:50:49.158 [update notice] received update-read broadcast outsidetemp QQ=10: 13.430
```

when querying the ebusctl, it doesn't look like it's recognizing what's on the ebus:
```
$ ebusctl info
version: ebusd 21.2.v21.2
update check: OK
signal: no signal
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 01: master #6, ebusd
address 06: slave #6, ebusd
```

```
~$ ebusctl read SensorData
ERR: element not found
```

Is this expected behaviour or might I be missing a step somewhere?

regards
Stijn

john30, 2021-03-12:
check if ebusd is running only once

goosst, 2021-03-12:
Not sure how I started it multiple times. Seems like after every reboot it starts to work until I start messing with the systemctl ... commands. Thanks for the support