john30/ebusd-esp #37: Reboot strategy if wifi connection is lost

tilman, 2021-04-06:
My esp-ebus-adapter sometimes becomes unavailable and does not come up again. From other projects I know that the low-level IP stack of esp sometimes can make problems and requires a full reboot to be able to connect to the wifi again. 
Therefore I wonder if there is any restart strategy implemented if the wifi connection is lost (or/and if no ebusd is connected) for a couple of minutes?

HeikoGr, 2021-04-06:
I used FHEM and had similar issues.

The ebus-esp came back to the wifi-network but FHEM or the esp has some trouble to reinit properly.

Until today in the morning i had a check in FHEM using doif and GetHttpFile (i don't use the esp Version since adapter v. 3 is out)

```
Define DoIf_ebus_bai_down DOIF
([MQTT2_ebusd_bai]) \
 ()\
 ({GetHttpFile("ebus-wifi.fritz.box", "/config?reboot=1")})\
 (set push msg 'ebus bai down')\
 (set push msg 'ebus bai still down')

attr DoIf_ebus_bai_down do resetwait
attr DoIf_ebus_bai_down room Intern->Push
attr DoIf_ebus_bai_down wait 0,120,240:0
```

so i forced a reboot if the ebus-esp went unreachable.

maybe this solves your problem or gives you a hint.

tilman, 2021-04-06:
Actually the esp disconnects from the wifi completely in my case. So it is not reachable via the network anymore (no ping response & not visible in the router UI).

john30, 2021-04-11:
does pressing the reset button help in this case?

tilman, 2021-04-11:
Yes, pressing the reset button works as well.

john30, 2021-04-11:
note to myself: perform hardest possible reset if there is no wifi connection for more than a certain time

tilman, 2022-08-21:
Hey, are there any updates on this? :)

john30, 2022-10-30:
not yet

kecksk0enig, 2023-10-10:
I recently upgraded to latest firmware, since then I get wifi Disconnects way more often.the installation/position of AP and ebus adapter did not change.

could there be a flaw in the current firmware resulting in wifi disconnect?
Currently  I have to reset the adapter everey 2 to three hours.

Is there any way to provide useful logs concerning this issue?