john30/ebusd-esp #38: Pedotherm / Vaillant - recoVAIR 350 - connection?

marthinx, 2021-04-08:
Thank you very much for your your code!

I am using a nodeMCU in combination with the Pedotherm / Vaillant - recoVAIR 350.

I think I have a working ebus connection.

Here is the status:

eBUS adapter 3
Build: 20201122
ebusd device string: enh:192.168.178.126:9999
WiFi signal: 54% (-73dBm)
Status
Configuration
PINs
Firmware update
Chip ID: 0058bb2d
Hostname: ebusLueftung
Up time: 204
Free heap: 34288
ebusd connected: no
eBUS signal: acquired
eBUS initial signal: yes

But I can not see any logs here:
http://192.168.178.126:9999/

I alway get the error message:
ERR_INVALID_HTTP_RESPONSE

What can I do? Who can I get the data or the control for this device?

Thank you very much,

Regards,

Martin

HeikoGr, 2021-04-09:
The ebusd-esp firmware has no http/html Output on Port 9999. You need the ebusd Software running and connected to the ebus-esp. 

On this image you can see the concept behind heating-device <> ebus-adapter <> ebusd (daemon) <> home-automation
[image]

john30, 2021-04-11:
exactly, thanks @HeikoGr.
you need to have an ebusd instance running somewhere that interprets the eBUS traffic. or something else that can handle that

closed 2021-04-11 by john30