john30/ebusd-esp #40: Gas sensor support on the PCB (5V ??) + software questions

netweaver1970, 2021-04-19:
Hi, 
I want to use the EBUSD v3 adapter together with a MQ5 gas sensor. 
The MQ5 module needs 5V. 

I see there's a J3 header proving GND + D7 pins. Since that D7 pin is directly connected to the Wemos input pin, do we have to use a voltage divider to feed the 3.3v high level through?

I'm powering the EBUSD v3 adapter using USB port J2, so as far as I can see (in the schematic), there's no 5V pin easily accessible anymore on the board. Am I correct?
Where would you then get the required 5V from? Apart from running a tiny wire from the input leg of the TS1117 LDO underneath the Wemos D1 mini? 
That should not be the preferred/only way...

Or did you have another gas-sensor in mind when designing this board?

Also, how would the binary gas/no gas detected state be exposed in software? Can it publish MQTT messages? Any examples?

Overall, I see there's a definitive idea to support a gas sensor on the PCB and in the web config frontend but the actual implementation/usage is unclear from the documentation.

Also, I would like to use ESBHome as sensor platform for the EBUSD module to expose to HomeAssistant. I see in the documentation that these things are [todo]. Are they still coming or is this due to real-life issues (aka lack of time) in-definitively postponed?

Thanks.

netweaver1970, 2021-04-19:
answering my own question on the input pin level shifting, it seems the GPIO pins are 5V tolerant but can sink only up to 20-25mA. so a 10K resistor in series would be more than enough to protect the pin, if one really wanted.

The other questions still stand though :)

netweaver1970, 2021-04-21:
I decide to jump ahead. I did as I planned, soldered the MQ5 module's 5V supply line on the IN pin of the TS1117 and took GND and D7 from J3, to bring to the MQ5 gas sensor module. I can see the gas sensor is nicely working, when testing with lighter gas, I can see the DO line of the module go from 'H' to 'L' and the green 'gas detected' LED goes on. And I can see the Wemos also picks up the pin state correctly in the web PIN config screen. I'm still not convinced this is the designed way so I'm leaving this open for comments.

netweaver1970, 2021-04-27:
@John30 Now that the adapter is finally providing proper heater messages via mqtt, how can I get the state of the other wemos input pins, such as the gas/switch input? And how to set the pin output levels? And interact via i2c / single wire devices?

john30, 2021-06-06:
for the time being you can only use the JSON format to interact with the PINs and for now this is only for on/off and analog read, as well as I2C temp sensors configured in the UI

netweaver1970, 2022-11-04:
Project stalled, other prios came in. Also ebus adapter is not responding anymore, needs reflash? Tbc later.

closed 2022-11-04 by netweaver1970