john30/ebusd-esp #44: Can't save configuration

mkrzywie, 2021-04-28:
I've problem with enabling eBus server on my WeeMos D1 mini.
Whenever I'm trying to save config and reboot I'm getting to default state with EBUS AP and following DHC status:
WIFI IP address: DHCP (failed: disconnected, reason: 1)

Config steps:
1. Choose SSID (option 1)
2. Set WiFi secret (option 2)
3. Check connection (option c). After this step I see device got IP address and I can log in on device.
4. Save and reset (option 0)

After that all time I getting:

load 0x4010f000, len 3584, room 16
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

   Welcome to eBUS adapter 3, build 20201122
                                            Configured as WIFI access point EBUS without password.
                                                                                                  For configuration with web browser, connect to this WIFI and open http://192.168.4.1/
                                                                Entering configuration mode.
                                                                                            Chip ID: 00458AE4
                                                                                                             Hostname: ebus-******

           Configuration (new):
                                1. WIFI SSID: **EBUS**
                                                   2. WIFI secret:
                                                                    3. WIFI IP address: **DHCP (failed: disconnected, reason: 1)**
        w. WIFI power: normal
                              4. WIFI hostname: ebus-458ae4
                                                            5. eBUS RX+TX PINs: Adapter 3 RX+TX (GPIO3+1)
                                                                                                          6. ebusd connection: enhanced on port 9999
                              7. HTTP TCP port: 80
                                                   8. LED PINs: RX:disabled, TX:disabled
                                                                                         9. Initial PINs: D4:H

                                                                                                               p. Set current PINs: D4:H
                  t. Toggle current output PIN
                                               s. Scan & read sensors
                                                                      c. Connect WIFI
                                                                                      e. Dump EEPROM content
                                                                                                             f. Load factory settings
               F. Factory reset (i.e. erase EEPROM)
                                                    o. OTA enabled: waiting
                                                                            r. Reboot (without saving)
                                                                                                       0. Save configuration and reboot

                Enter your choice:

netweaver1970, 2021-04-29:
I had the same issue, I could only solve it by reflashing the Wemos d1 mini. 

During reflash a full flash clear was done and afterwards I could save my config.

I got to that idea by seeing a rather different (simulated) eeprom dump before and after a save+reboot.

mkrzywie, 2021-05-19:
Thanks,

In my case reflashing with ebus-v3_d1mini_lite helped.

closed 2021-05-19 by mkrzywie