john30/ebusd-esp #45: OpenSource like ebusd?

kodidu, 2021-05-14:
Hello!
I could not find the source code for the binaries in the dist folder.
I would like to port it over to STM32F4 with integrated ethernet MAC.

kodidu, 2021-06-07:
@john30 Any update on this topic?
I want to use wired Ethernet instead of WLAN.

kodidu, 2021-07-25:
Project dead?

john30, 2021-10-03:
see #5

jmenoremont, 2021-10-13:
Those closed sources are worthless...reinvent wheel or reverse it

closed 2021-10-30 by john30

kodidu, 2021-11-19:
> @john30 @jmenoremont 
> see #5

Ok, I am sure the decision to make money by selling hardware will also affect ebusd.
I will move away from the whole ebusd project.
I won't miss the strange configuration csv files with inheritance :grin:. You never know in which column you are...

john30, 2021-11-19:
wow, such comments really make me think: why the hell am I spending tons of my free time to do all these projects?
please explain: how does one make money with providing hardware that costs just as much as the price sold for?
and what's so complicated about writing yourself a simple Wifi-serial bridge on an ESP? just do it!
anyway, I will restart this project from scratch based on RIOT. I am looking forward to your contributions (but I guess you won't do so anyway).

kodidu, 2021-11-20:
>please explain: how does one make money with providing hardware that costs just as much as the price sold for?

I can't know that. For me making money seams to be the only reason for storing binaries in a public repository without the source. 

> and what's so complicated about writing yourself a simple Wifi-serial bridge on an ESP? 

Ah ok, now I get it. The ESP has nothing to do with the "enhanced ebusd protocol".
There was some misunderstanding. I am not after the ESP source, but after the source of the PIC micro controller firmware.
The PIC does the time sensitive ebus decoding, buffers the ebus messages and sends "enhanced ebusd protocol" frames via the serial Wifi ESP bridge to ebusd running on Linux. This makes sense because real time tty communication with Linux is not an easy task without rt scheduler patch.

I do not want to use Wifi for fixed installations.
My main intention was to port it over to STM32Fx with wired Ethernet. This could combine the PIC and ESP in a single device.

Now the question is: Will the PIC firmware become opensource like the schematics?

john30, 2021-11-25:
the PIC also supports using a wired connection via Ethernet, [see here](https://adapter.ebusd.eu/index.en.html#ethernet).
What do you want to do with firmware for a PIC16 that certainly does not fit at all to a STM32?

kodidu, 2021-12-05:
Ok, with these wiznet chips. Long time ago I used the enc28j60 with the atmel megas. enc28j60 uses a lot of power in idle.
How is the power consumption with these older chips?

I like to tinker around with the stuff on hand instead of buying. I already built my own usb -> ttl -> ebus converter using stm32.
After seeing the new ebus adapter it would make sense to also improve my adapter design.
Porting should not be a hassle if the software is kept simple. Maybe only the Hardware peripherals UART, TIMER,... must be ported.
Looks like the ebus is decoded in software using the dedicated compare unit. On stm32 i can do this using the adc and analog watchdog. But maybe this software defined thresholds are only a nice to have and an external comparator with preset thresholds is enough like in the old schematics.

john30, 2021-12-12:
W5500 power consumption is within specifications, below 200 mA

kodidu, 2021-12-13:
ok.
As it stands, no plans to release the PIC firmware?

xyzroe, 2021-12-20:
Greetings!

I am trying to make my own ebus gateway because:
1. I need LAN connection and POE power supply
2. I can't buy from you, because you don't have the parts you need.
I guess the problem is in the PIC16F15356, it is out of stock.

But I found some on taobao, but that won't save the day for long.

Perhaps open source code would allow building firmware for those chips that are available for purchase now.

I am planning to use the LILYGO TTGO T-Internet-POE ESP32 LAN8270A board by making a shield for it.
The source code of the firmware for the ESP and Gerbers of board will be publicly available.