john30/ebusd-esp #46: Configuration not saving D1 mini PRO

MagMeister, 2021-05-28:
I bought this D1 mini Pro with external antenna to improve the WiFi range of my ebus adapter 3:
https://www.reichelt.com/be/nl/d1-mini-pro-esp8266-cp2104-set-met-antenne-d1-mini-pro-ant-p266067.html

After flashing with the [Pro firmware](https://github.com/john30/ebusd-esp/blob/master/dist/ebus-v3_d1mini_pro.bin), I can connect to the EBUS access point and use the web interface at 192.168.4.1. But when I change the configuration to use my home WiFi access point, check & load, save & reboot, the device reboots but the configuration was not stored. The EBUS access point comes up again.

I found this [closed issue](https://github.com/john30/ebusd-esp/issues/23), but I couldn't find any useful information.

MagMeister, 2021-05-29:
Update: I tried using the serial interface to configure the WiFi settings. No success, same issue, configuration is not stored.

Then I flashed the device with the "normal" D1 mini ebus-v3_d1mini.bin firmware. I was able to successfully configure the WiFi settings. Unfortunately the WiFi performance with external antenna is worse than before when using a regular D1 mini. But maybe that's because the firmware doesn't comply for this?

Anyway, the Pro firmware is not working for me.

MagMeister, 2021-05-29:
Again an update. About the WiFi performance: stupid me, I had to re-solder the jumper as described in the specification to use an external antenna instead of the internal. Lessons learned: the internal antenna of the pro version performs worse than the antenna on the regular version.

Everything fine for now. I will keep using the D1 mini ebus-v3_d1mini.bin firmware on my D1 mini Pro and see if it keeps stable.

john30, 2021-10-30:
can you try with the latest firmware again please?

mbisak, 2021-11-30:
I have downloaded firmware an hour ago and suffer from the same issue. Cannot save config to D1 mini PRO with PRO firmware.

mbisak, 2021-11-30:
With D1 mini (normal) Wemos load config, with PRO, it does not. Version 20211030

mbisak, 2021-11-30:
One more observation, configuration is stored correctly but PRO firmware does not load it after boot. When I just flash ESP with different firmware, without changing anything, it configures itself ok with normal version and does not with PRO.

Martin

john30, 2021-12-12:
is this still the case with the latest version?
I don't have a pro wemos, so I can't test it. if the other firmware works fine, then use that one

mbisak, 2021-12-12:
> Odesláno: neděle 12. prosince 2021 19:31
> Komu: john30/ebusd-esp

Still the same. I use version for normal esp version and it s OK.

Martin

wairlex, 2021-12-21:
Same issue for me. I have seen that from the serial interface, there is an option to dump eeprom content. Is the configuration  stored in the eeprom? Can we help you to try to find where the bug is?

john30, 2021-12-23:
then just stick to the non-"pro" version for now. I need to get one myself

john30, 2021-12-26:
meanwhile I found out that many of the "pro" versions sold are actually equipped with only 4 MB of flash instead of 16 MB as stated on the Wemos product page.
Consequently, the non-pro image needs to be flashed on those devices. I updated the [readme ](https://github.com/john30/ebusd-esp/blob/master/README.md) to make that clear.

closed 2021-12-26 by john30

pepsonEL, 2022-01-08:
Hi> I buy wemos d1 mini pro with 16MB flash. Seller tell me that has 16MB Flash. I flash it by version firmware PRO 16MB 20211205 and also has issue that not save settings. After reboot again come back to access point EBUS.

pepsonEL, 2022-01-08:
I flashed by ESPHome-Flasher-1.2.0-windows-x64 and now settings are saved