john30/ebusd-esp #47: Flashing instructions for ESP32

tmshlvck, 2021-08-19:
The README lacks flashing instructions for ESP32. I tried this to flash the Lolin32 binary to ESP32-DevKitC board:

```
th@slepice:~/ebusd-esp/dist$ ~/.local/bin/esptool.py --port /dev/ttyUSB0 erase_flash
esptool.py v3.1
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, Coding Scheme None
Crystal is 40MHz
MAC: 24:0a:c4:13:3e:58
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 7.3s
Hard resetting via RTS pin...

th@slepice:~/ebusd-esp/dist$ esptool.py --port /dev/ttyUSB0 write_flash -z 0x1000 ebus-v3_lolin32.bin 
esptool.py v3.1
Serial port /dev/ttyUSB0
Connecting.......
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, Coding Scheme None
Crystal is 40MHz
MAC: 24:0a:c4:13:3e:58
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x000d0fff...
Compressed 849472 bytes to 477659...
Wrote 849472 bytes (477659 compressed) at 0x00001000 in 42.5 seconds (effective 159.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

After reset I am getting reboot loop:
```
th@slepice:~/ebusd-esp/dist$ screen /dev/ttyUSB0 115200

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3f400020,len:192468
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3f400020,len:192468
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3f400020,len:192468
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3f400020,len:192468
ets Jun  8 2016 00:22:57
```

I am using this version of esptool:
```
th@slepice:~/ebusd-esp/dist$ ~/.local/bin/esptool.py version
esptool.py v3.1
3.1
```

closed 2021-10-30 by john30

meyerd, 2022-04-14:
i used 

`esptool --chip esp32 -p /dev/ttyUSB0 -b 921600 write_flash 0x00000 ebus-v3_lolin32_factory.bin`

(base address for flash 0x0000) which worked

viktor-c, 2022-08-24:
I wanted to try the bin image out
using 
`esptool --chip esp32 -p /dev/ttyUSB0 -b 921600 write_flash 0x0 ebus-v3_lolin32_factory.bin`
writes to esp32 with following log
`esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: xx:xx:xx:xx:xx:xx
Changing baud rate to 921600
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Auto-detected Flash size: 4MB
Erasing flash...
Took 0.88s to erase flash block
Wrote 135168 bytes at 0x00000000 in 2.5 seconds (431.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...`

gives me following error in the serial console
`rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57`

What I am wondering, is why the latest image is 131kB and the mentioned above is almost 850kB

john30, 2022-09-17:
maybe you want to have a look at the [flash.sh script](https://github.com/john30/ebusd-esp/blob/master/flash.sh)

commit 79ecee3f (2021-10-30): add factory image for ESP32 (fixes #47)