john30/ebusd-esp #49: no eBUS signal with Protherm Skat

hryamzik, 2021-09-19:
Hardware:

`NodeMcu v1.0`
`Protherm Skat 9`

Using all the defaults and RX/TX pins I get the following status:

```
Chip ID: 007a55a0
Hostname: ebus-7a55a0
Up time: 72
Free heap: 34736
ebusd connected: no
eBUS signal: no signal
eBUS initial signal: echo
```

And heater fails to `f.49` error, according to manuals description is `Short circuit on the eBus, overload on the eBus or on the eBus 2 of a voltage source with different polarity`.

Any ideas how to debug this?

john30, 2021-10-03:
yes, you need an [adapter like this](https://adapter.ebusd.eu/index.en.html) to gain access to the bus signal

hryamzik, 2021-10-04:
Thanks, I've already placed my order!

closed 2021-10-04 by hryamzik