john30/ebusd-esp #53: Adapter 3 wemos no eBus signal

isvedic, 2021-11-13:
Adapter 3 with wemos WiFi and all connected with default settings, I'm having issues, even with ebus wired and green LED flashing and jumpers set as requested for WiFi wemos still reports eBUS signal: no signal with GPIO3+1 selected for Adapter3

For some time there was under status eBus initial signal: YES but that is gone now.

john30, 2021-11-14:
did you set the device string on ebusd correctly?

isvedic, 2021-11-14:
Not sure about device string since I'm using it under home assistant, ebusd connection from there looks ok and wemos reports ebusd connected and get blue light on it.

john30, 2021-11-16:
there should be an ebusd running and it needs to know about the device to connect to

Koky05, 2021-11-26:
Adapter 3 with wemos WiFi and all connected with default settings.
My configuration for HA is:
`ebusd:
  host: 192.168.2.31
  port: 9999
  circuit: "ehp"`

But I alway see only:
Build: 20211030
ebusd device string: enh:192.168.2.31:9999
WiFi signal: 62% (-69dBm)
OTA enabled: waiting
Chip ID: 0018f4ba
Hostname: ebus-18f4ba
Up time: 796
Free heap: 28584
ebusd connected: no
eBUS signal: no signal
eBUS initial signal: yes

Where I made a mistake?

john30, 2021-11-28:
what does the ebusd log say?

Koky05, 2021-11-28:
> what does the ebusd log say?

I found that I did not install any ebused because I was thinking it is in wemos firmware or Home Assistat it contains. But I need to install your ebusd into docker (I run HA on Synology NAS also in docker).

Koky05, 2021-11-29:
> > what does the ebusd log say?
> 
> ....
I install ebusd in docker with next command:
`docker run --name=ebusd -d --restart=unless-stopped --net=host -p 8888:8888 -e TZ=Europe/Bratislava -v /volume1/docker/ebusd/data:/data --label=com.centurylinklabs.watchtower.enable=true john30/ebusd -f --scanconfig -d enh:192.168.2.31:9999 --latency=2000 --loglevel debug`

And in log I found:
`re-opened 192.168.2.31:9999
[main debug] performing regular tasks
[bus notice] device invalid`

Koky05, 2021-12-02:
This is picture from MQTT Explorer and show all the data I can actualy read from my heating system. In this state it is useless and I lost my money that I pay for that adapter. I did not see actual configuration no actual temperature (only outside), etc...
[image]

john30, 2021-12-05:
nothing strange to see here. I don't know what you expect. Have a look at the [wiki](https://github.com/john30/ebusd/wiki) and don't ask me for any home assistant stuff, as that's not my project.

Koky05, 2021-12-05:
@john30 sorry for bithering you. I was expected to see all variables after first load not only some. And because I have VR61 to control independent two heating circuite it is shame to nit to.see same ingormation for VF2 like I can see and setup for VF1. But I understand niw that it is limitation of software.

john30, 2021-12-05:
rather a limitation of the user. you need to query for most of the values actively and not just wait for them to arrive

Koky05, 2021-12-05:
@john30 how can I make same file for 4700, because full scan it found on 15 qnd also on 26. But if I just make new file 26.470.csv it just show error about duplicity of configuration file.

john30, 2021-12-23:
why do you want to copy it?

Koky05, 2021-12-23:
@john30 Because I want to see what type of data is there and why it is twice. But I found solution. Just creted link as new file 26.470.2.csv from file 15.470.csv.

closed 2021-12-26 by john30