john30/ebusd-esp #54: ebus adapter read consuption

pepsonEL, 2021-11-15:
Hi all… I want use ebus adapter 3 on wifi with my boiler GAS Vaillant VSC206 4-5. Is any user which use it with Vaillant Boiler ? I have with my vaillant VRC700 Multimatic and it show gas and energy consuption.
But i have also Vaillant Gateway VR920 but in app Android Vaillant not show consuption… only show in Multimatic VRC700. Is any chance to read consuption by ebus adapter 3 ? Please help me thanks.

kalledausb, 2021-11-15:
Hi, I also don't have the Value in the Vaillant App, according to Vaillant support it is only available for ectotec exclusive devices as in the others the value is not very reliable.
Anyway You can query it with ebusd, PrFuelSum for example might be the data point to look at.

pepsonEL, 2021-11-15:
Ok but i think about buy adapter and i dont know if i can read it... Good will be if someone can verify it.... In VRC700 show me consuption....

kalledausb, 2021-11-15:
I can confirm that I also have the values on VRC700 display and also that in ebusd I do capture them for long term analysis. However I do not know (yet) how accurate they are

pepsonEL, 2021-11-15:
Ok but tell me how add it by ebus to HA ?

kalledausb, 2021-11-15:
What is HA?

pepsonEL, 2021-11-15:
Home Assistant

kalledausb, 2021-11-15:
OK, then I am afraid I can't help further - I use ebusd only in combination with ioBroker to collect needed values frequently and store in a database then. Let's see if somebody else can

john30, 2021-11-16:
this is the wrong repo for such questions, please use [ebusd-configuration](https://github.com/john30/ebusd-configuration) instead.
anyway, you need to look for available message definitions ins ebusd when it was running for a couple of minutes. only if it is available there, there is a chance to get it on to an integration

closed 2021-11-16 by john30

pepsonEL, 2021-11-16:
ok thanks.
BTW.
I create reservation for ebus adapter 3 but universal wifi and LAN. I get email to pay for it but i change it and now i want only with wifi. How i can change it and how i can pay for it ? And when did you plan start sending it ?

pepsonEL, 2021-11-16:
And is any chance to run it with your Adapter ebus 3 on Home Assistant with installed on SSD in HP Terminal T630 HASSOS (image system) ?

pepsonEL, 2021-11-19:
Hi... how i can change my reservation from universal to only with wifi ? ANd how i caan pay.... Is any conntact with you ?

john30, 2021-11-20:
> Hi... how i can change my reservation from universal to only with wifi ? ANd how i caan pay.... Is any conntact with you ?

you could have just replied to one of the emails that were sent to you. or maybe your address is categorized as spam. anyway, with the universal version you get wifi as well, so what's the matter?

pepsonEL, 2021-11-25:
I reply email but with no any answer....

john30, 2021-11-30:
again: with the Universal variant you get all options, so why do you want to change it?
I've asked the colleague to check for your mails. A hint to your mail address (at least parts of it) would be helpful.
Anyway, I'm not sure if your adapter is still available since quite some time has passed now and we usually give it away to the next on the list if it is not payed within a week (as stated in the mail)

pepsonEL, 2021-11-30:
Because price is lower.... 
My email is piotr.kucxxxxxxx@gmail.com

pepsonEL, 2021-11-30:
And only please tell me i run this adapter on HASSOS  to comunicate HA with my boiler gas Vaillant ? But i have installed on my terminal HP T630 image with HASSOS.

john30, 2021-11-30:
wow, the whole time consuming discussion for saving a couple of euros, great.
I have no idea, how I should answer your second comment. I don't understand

pepsonEL, 2021-11-30:
I must only write firmware to wemos and then connect to wifi and in HA add in configuration ebus platform ?

john30, 2021-12-05:
I don't know. ask HA for that