john30/ebusd-esp #55: Adapter 3 + wemos d1 mini pro -ebusd connected but no signal

sebastiangebosz, 2021-11-28:
Hi.

I have a problem with NO-SGINAL...

Sounvier Duval Thelia Condens Piec AS 25-A + EXACONTROL E 7 C.

Adapter: eBUS Adapter 3 + Wemos D1 mini PRO (flashed by ebus-v3_d1mini.bin, becouse ebus-v3_d1mini_pro.bin cant save settings)

Ebusd runs on Home Assistant Portainer - Docker image from Docker.io.
Container run with CMD: -f --scanconfig -d enh:192.168.1.10:9999 --latency=80

Wemos d1 mini pro:
STATUS:
Build: 20211030
ebusd device string: enh:192.168.1.10:9999
WiFi signal: 64% (-68dBm)

Chip ID: 0092523a
Hostname: ebus-92523a
Up time: 188
Free heap: 31208
ebusd connected: yes (inactive)
eBUS signal: no signal 

COFIGURATION:
ebus RX+TX Pins : Adapter 3 RX+TX(GPIO3+1)
ebus connection: enhaced on port 9999
HTTP TCP port 80

CONTAINER LOG:
2021-11-28 10:38:38.102 [main notice] ebusd 21.3.v21.3 started with auto scan on enhanced device 192.168.1.10:9999
,2021-11-28 10:38:45.050 [bus notice] device status: resetting
,2021-11-28 10:38:45.051 [bus notice] bus started with own address 31/36
,2021-11-28 10:40:55.231 [main notice] update check: version 21.2 available

If I change in wemos configuration 
ebus RX+TX Pins : to direct 3 RX+TX(GPIO3+1)

CONTAINER LOG shows:
2021-11-28 10:44:11.651 [main notice] ebusd 21.3.v21.3 started with auto scan on enhanced device 192.168.1.10:9999
,2021-11-28 10:44:16.977 [main error] error reading config files from http://ebusd.eu/config/: ERR: element not found, last error: 
,2021-11-28 10:44:17.177 [bus notice] device status: resetting
,2021-11-28 10:44:17.178 [bus notice] bus started with own address 31/36
,2021-11-28 10:44:17.260 [bus notice] device status: reset
,2021-11-28 10:44:17.301 [bus notice] device status: reset

The jumpers are set as discribed on webside:
J1 RPI, 
J4 RPI, 
J12 - 4-5

Green led on adapter the green LED flashes as if it is communicating with the boiler..
What im doing wrong?

john30, 2021-11-30:
how do you power the adapter?

sebastiangebosz, 2021-12-01:
I have made progress. I bought a new wemos. This time a normal mini (not pro). I have uploaded dedicated software. I also replaced the charger and power cord.

The jumpers are set as discribed on webside:
J1 RPI,
J4 RPI,
J12 - 4-5

Green led on adapter the green LED flashes
Red led flashes sometimes.

Container run with CMD:   -f --scanconfig -d enh:192.168.1.10:9999 --latency=200

CONTAINER LOG:
2021-12-01 17:22:51.183 [main notice] ebusd 21.3.v21.3 started with auto scan on enhanced device 192.168.1.10:9999
,2021-12-01 17:22:52.215 [bus notice] device status: resetting
,2021-12-01 17:22:52.217 [bus notice] bus started with own address 31/36
,2021-12-01 17:22:52.503 [bus notice] signal acquired
,2021-12-01 17:22:52.652 [bus notice] device status: reset
,2021-12-01 17:22:56.264 [bus error] signal lost
,2021-12-01 17:22:57.375 [bus notice] signal acquired
,2021-12-01 17:22:57.538 [bus error] device status: unexpected available enhanced byte 2
,2021-12-01 17:23:02.261 [bus error] signal lost
,2021-12-01 17:23:03.215 [bus notice] signal acquired
,2021-12-01 17:23:03.375 [bus error] device status: unexpected available enhanced following byte 1
,2021-12-01 17:23:03.375 [bus error] device status: unexpected available enhanced following byte 1
,2021-12-01 17:23:04.126 [bus error] device status: unexpected available enhanced byte 2
,2021-12-01 17:23:08.232 [bus notice] new master 10, master count 2
,2021-12-01 17:23:08.599 [bus notice] new master 03, master count 3
,2021-12-01 17:23:08.599 [update notice] received unknown MS cmd: 1008b5110101 / 09181a0080ffff0000ff
,2021-12-01 17:23:10.315 [update notice] received unknown BC cmd: 10feb516080032221830110221
,2021-12-01 17:23:10.716 [update notice] received unknown MS cmd: 1008b5070108 / 026400
,2021-12-01 17:23:13.286 [update notice] received unknown MS cmd: 1008b51009000000ffffff050100 / 0101
,2021-12-01 17:23:15.956 [main error] scan config 08: ERR: SYN received
,2021-12-01 17:23:17.888 [update notice] received unknown MS cmd: 1008b5110101 / 09181a0080ffff0000ff
,2021-12-01 17:23:21.619 [bus error] device status: unexpected available enhanced byte 2
,2021-12-01 17:23:23.257 [main error] scan config 15: ERR: SYN received
,2021-12-01 17:23:24.261 [update notice] received unknown MS cmd: 1008b51009000000ffffff050100 / 0101
,2021-12-01 17:23:25.504 [bus error] device status: unexpected available enhanced byte 2
,2021-12-01 17:23:25.555 [bus error] device status: unexpected available enhanced byte 2
,2021-12-01 17:23:25.905 [bus error] device status: unexpected available enhanced byte 2
,2021-12-01 17:23:25.984 [bus error] device status: unexpected available enhanced byte 2
,2021-12-01 17:23:26.237 [bus error] device status: unexpected available enhanced byte 2
,2021-12-01 17:23:26.259 [bus error] device status: unexpected available enhanced byte 2
,2021-12-01 17:23:28.175 [bus error] signal lost
,2021-12-01 17:23:29.323 [bus notice] signal acquired
,2021-12-01 17:23:30.710 [bus error] device status: unexpected available enhanced following byte 1
,2021-12-01 17:23:31.525 [update notice] received unknown MS cmd: 1008b512020000 / 00
,2021-12-01 17:23:32.144 [bus notice] new master 00, master count 4
,2021-12-01 17:23:32.276 [bus error] device status: unexpected available enhanced following byte 1
,2021-12-01 17:23:34.321 [bus error] device status: unexpected available enhanced byte 2
,2021-12-01 17:23:34.568 [bus error] device status: unexpected available enhanced following byte 1
,2021-12-01 17:23:38.742 [bus error] device status: unexpected available enhanced following byte 1
,2021-12-01 17:23:39.928 [bus error] device status: unexpected available enhanced following byte 1
,2021-12-01 17:23:40.695 [main error] scan config 05: ERR: SYN received
,2021-12-01 17:23:41.307 [bus error] device status: unexpected available enhanced following byte 1
,2021-12-01 17:23:41.308 [bus error] device status: unexpected available enhanced following byte 1
,2021-12-01 17:23:42.228 [bus error] device status: unexpected available enhanced following byte 1
,2021-12-01 17:23:45.677 [bus error] device status: unexpected available enhanced following byte 1
,2021-12-01 17:23:46.159 [bus error] device status: unexpected available enhanced following byte 1
,2021-12-01 17:23:46.865 [bus error] device status: unexpected available enhanced byte 2
,2021-12-01 17:23:46.871 [bus error] device status: unexpected available enhanced byte 2
,2021-12-01 17:23:47.162 [bus error] device status: unexpected available enhanced byte 2
,2021-12-01 17:23:47.162 [bus error] device status: unexpected available enhanced byte 2
,2021-12-01 17:23:47.321 [bus error] device status: unexpected available enhanced byte 2
,2021-12-01 17:23:48.727 [bus error] device status: unexpected available enhanced following byte 1
,2021-12-01 17:23:49.172 [bus error] device status: unexpected available enhanced byte 2
,2021-12-01 17:23:49.733 [bus error] device status: unexpected available enhanced byte 2
,2021-12-01 17:23:50.043 [bus error] device status: unexpected available enhanced byte 2
,2021-12-01 17:23:50.351 [bus error] device status: unexpected available enhanced following byte 1

Koky05, 2021-12-02:
@sebastiangebosz I had same problem. Just do couple of restarts and it start normal. But then looks like my Vaillant setup was not correctly identified and only few data was received.

john30, 2021-12-05:
did you set the esp to Adapter 3 RX+TX (GPIO3+1) and enhanced?
If so, please flash the latest firmware from today

sebastiangebosz, 2021-12-13:
Yes, it was set up properly.

I tried the new soft.
Wemos mini does not show hotspot after fleshing.
Wemos mini pro does not save the settings.

I bought a raspberry 3. I installed a docker there. I replaced the cable from the boiler with a different one. Then I connected via ethernet adapter and it works.

Probably not read via docker in Portainer in Home Assistant. I do not know why..

john30, 2021-12-23:
in some other ticket it was said the the non-"pro" firmware works fine, maybe you want to try that

john30, 2021-12-26:
regarding the so called "pro" versions, please see my [comment here](https://github.com/john30/ebusd-esp/issues/46#issuecomment-1001190704)

pepsonEL, 2022-01-09:
I have the same issue. I have correct wemos d1 mini pro with 16MB flash. I write firmware for version pro. ANd now when i reboot ebus adapter 3 with wemos signal is lost and then is now signal. I must power off and again power ON and then signal is correct. But when again reboot from web page again signal is NO. Again replug POWER and again signal is YES. 

On wemos d1 mini is no this problem. On every reboot signal is YES.

andrejs2, 2022-02-17:
I have the same problem. D1 mini (not pro). 
The connection is some hours and then is signal lost (usually during the night). I could troubleshoot if only signal or mqtt connection with broker as well. After reseting (physical on wemos d1) the adapter establishes signal and everything is good for some time (hours) after it again loses the signal.

[image]

pepsonEL, 2022-02-17:
I have the problem that sometimes when i reboot adapter from web page , and after rebooted is no signal. Then i must power OFF adapter and again POWER ON and start works ok. 
I tested it also on WEMOS D1 MINI PRO(with verify 16MB Flash) and problem is the same.

john30, 2022-03-19:
what do the LEDs on the adapter look like in this situation?
and please try to reproduce with latest PIC firmware and latest ebusd-esp and latest ebusd

pepsonEL, 2022-03-20:
Now i use Version 20220220, after reboot led on adapter:Orange LED power ON, GREEN LED blinking.... and BLUE LED on wemos POWER ON.

Tommorow i update PIC Firmware to Version 20220313 and test it.

pepsonEL, 2022-03-21:
I updated today PIC to version Version 20220313. Wemos also has latest version FW and ebusd also 22.2. 
After first power ON i have:
ebusd connected: yes
eBUS signal: acquired

After reboot wemos from GUI after reboot no signal:
ebusd connected: yes
eBUS signal: no signal

I test it on WEMOS D1 MINI PRO 16MB

john30, 2022-03-26:
so the blue LED on adapter is off permanently?

pepsonEL, 2022-03-26:
Yes.... 
When adapter boot after power on blue led on adapter blinking then no light... And when adapter works ok blue led sometimes blink

But when Adapter reboot from GUI webpage blue led on adapter never light on and never blink. Only blink green led and light on orange led.

john30, 2022-03-26:
the blue LED on the adapter does not blink when you reset the Wemos via web UI?

pepsonEL, 2022-03-26:
Yes no blink.
Also after reset no blink and no light.

john30, 2022-03-27:
please check with [new PIC firmware 20220327](https://adapter.ebusd.eu/picfirmware.en.html#versions)

pepsonEL, 2022-03-27:
ok tomorrow  i test it....
Good option to think about write software for Windows to we can update PIC Firmware. I dont have raspberry and i must please my frient to use it.... Option for Windows will be help for me....

Koky05, 2022-03-27:
I try WSL2 on my Windows 11 and compile ebuspicloader but it does not have driver for ebus adapter. So the next step will by Virtual machine with debian on it.

pepsonEL, 2022-03-27:
But on virtual machine works forward USB ?

Koky05, 2022-03-27:
On older versions of Windows I did not have any problem. But I do not know hove it will work on Win11 yet.

pepsonEL, 2022-03-28:
But what software use to virtualization ?

pepsonEL, 2022-03-28:
I try on Win10 with Hyper-V and installed as virtual machine Raspian OS... but he not see my adapter connected to my PC with Windows 10. In windows 10 it is found on COM5

Koky05, 2022-03-28:
https://adamtheautomator.com/hyper-v-usb-passthrough/

pepsonEL, 2022-03-28:
i downloaded iso Raspian OS from this link and create in VirtualBox Oracle 6.1 virtual machen with default option.
https://www.raspberrypi.com/software/raspberry-pi-desktop/

https://roboticsbackend.com/install-raspbian-desktop-on-a-virtual-machine-virtualbox/

pepsonEL, 2022-03-28:
On latest PIC FW looks that all works ok. Thanks for your work

netweaver1970, 2022-03-28:
Thanks for the fixing and the testing! I'll upgrade when back home too

Koky05, 2022-03-29:
After lots of problems with access rights on Windows 11 and on Virtual machine, I finally upload new PIC Firmware. It brings me more stability and zero needs of manual disconnecting of device. It will be great like @pepsonEL  mentioned to have also already compiled ebuspicloader for windows OS or option to upload FW through WiFi.

john30, 2022-04-10:
you can use socat.exe on windows and then run ebuspicloader in WSL, just tested it and it works fine

pepsonEL, 2022-04-11:
What is WSL?

kratz00, 2022-04-11:
@pepsonEL https://en.wikipedia.org/wiki/Windows_Subsystem_for_Linux

john30, 2022-04-16:
another alternative is usbipd being run on windows host which is capable of routing USB into the WSL