john30/ebusd-esp #56: No Signal

moskovskiy82, 2021-12-13:
Had an adapter 3 board with Wemos. Everthing was working correctly but had to replace the Wemos. Flashed with the mini firmware.

Now i get
ebusd connected: yes (inactive)
eBUS signal: no signal
eBUS initial signal: echo

And in the ebusd logs
2021-12-13 03:57:49.102 [bus error] device status: unexpected enhanced command 0x0e

But if i change the 
eBUS RX+TX PINs:  from Board3 to Direct

I get ebus signal acquired but ebusd throws this errors

2021-12-13 03:52:47.298 [bus info] scan 08 cmd: 3108070400
2021-12-13 03:52:47.571 [main error] scan config 08: ERR: no signal
2021-12-13 03:52:50.302 [main debug] performing regular tasks
2021-12-13 03:52:51.585 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-13 03:52:52.343 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-13 03:52:53.122 [main debug] performing regular tasks
2021-12-13 03:53:00.302 [main debug] performing regular tasks
2021-12-13 03:53:03.122 [main debug] performing regular tasks
2021-12-13 03:53:07.571 [main debug] performing regular tasks
2021-12-13 03:53:10.302 [main debug] performing regular tasks
2021-12-13 03:53:13.122 [main debug] performing regular tasks
2021-12-13 03:53:17.572 [main debug] performing regular tasks
2021-12-13 03:53:17.572 [bus info] scan 08 cmd: 3108070400
2021-12-13 03:53:17.844 [main error] scan config 08: ERR: no signal

moskovskiy82, 2021-12-16:
As an update Direct mode works just fine

john30, 2021-12-18:
are the jumpers set correctly?

moskovskiy82, 2021-12-29:
Had to switch to an USB connected to raspberry

closed 2021-12-29 by moskovskiy82