john30/ebusd-esp #59: OTA fails:  eBUS Adapter 3 with Wemos D1 mini

kratz00, 2021-12-29:
I am currently running build 20211205. 
I downloaded https://github.com/john30/ebusd-esp/blob/master/dist/ebus-v3_d1mini.bin and uploaded this file via the web front-end.
When the adapter comes back online it still show the same build number.

I would expect a different build number to be shown in the web front-end.

The file seems to be okay (also re-downloaded a couple of times)

```
[kratz00@manjaro _tv]$ strings /tmp/ebus-v3_d1mini.bin | grep 2021
Build: 20211226<br>
Build system: pio-espressif8266-arduino-d1_mini:2.2.0:6.15.2:@Dec 26 2021, 15:05:30<br>
Welcome to eBUS adapter 3, build 20211226
Build system: pio-espressif8266-arduino-d1_mini:2.2.0:6.15.2:@Dec 26 2021, 15:05:30
```

Any idea what is going on?
If not I will directly flash it, without OTA.

john30, 2021-12-31:
you can check on the console, sometimes the updater reports what the problem is, e.g. not enough memory (on 1M modules)

kratz00, 2022-01-02:
Serial output did not show much (output from the time I hit the *Upload* button):

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00060770
~ld

Welcome to eBUS adapter 3, build 20211205
```

I flashed the adapter with *esptool.py*, it is running build *20211226* now and everything seems fine so far.

closed 2022-01-02 by kratz00