john30/ebusd-esp #60: Possible to use ebusd-esp as a bus monitor to log communications between other devices on the bus?

fhoguin, 2022-01-09:
Hello,

I would like to spy or monitor communications on an eBUS system. I was wondering if ebusd-esp can be used for that? It would be useful to see what messages the control panel sends on the bus when e.g. pushing a button, in order to analyze and replicate them.

From what I can gather from the doc, I assume the ebusd-esp device acts as a master (and/or slave?) on the bus, and will therefore ignore messages not adressed to it.

I tried to find the firmware's source code to confirm, but it seems it is not available, is that correct?

Thanks in advance for any answer.

kodidu, 2022-01-14:
> I tried to find the firmware's source code to confirm, but it seems it is not available, is that correct?
See also #45, sadly there is no clear statement about the source code.

john30, 2022-01-26:
ebusd-esp is nothing more than a serial-wifi bridge. no mast/slave mode or whatsoever