john30/ebusd-esp #61: Software License and source code access

mlesniew, 2022-01-18:
Hi, 

I couldn't find license information in this repository.  Would you mind adding a license file, similar to how it's done in the [ebusd repo](https://github.com/john30/ebusd)?  This would be useful to clarify how others can use it.

Even if you haven't decided what license to use, is it fair to say that this is open software?  If so, how can source code be accessed?

Cheers

mlesniew, 2022-01-18:
Looking around the repo I discovered that it uses the [EspSoftwareSerial library](https://github.com/plerup/espsoftwareserial), which is LGPL licensed.   I think this limits the license choice to just LGPL and anything LGPL-compatible.

HeikoGr, 2022-01-18:
you are wrong:

```The LGPL is a weaker form of the GPL that is not contagious over library boundaries. Originally, LGPL was read out as Library General Public License. It has now been reinterpreted to be read out as the Lesser General Public License. I assume they mean lesser, as in less strong in forcing software to be open source.

Applying LGPL to a library ensures that the library itself and any modified versions of it will remain open source. But it can be used by closed source software.```
(Source: https://dzone.com/articles/the-lgpl-license)

mlesniew, 2022-01-18:
LGPL is less contagious than GPL, but the rule about not crossing library boundaries only holds for code which is linked dynamically.  This is widely discussed online, it's also explained in the [GNU License FAQ](https://www.gnu.org/licenses/gpl-faq.en.html#LGPLStaticVsDynamic).

In this project, the EspSoftwareSerial library is linked statically (as usual in the microcontroller world). 

Just because of this static vs dynamic linking issue, many projects use a LGPL license with an additional exception about static linking, e.g. [wxWidgets](https://www.wxwidgets.org/about/licence/) or [OpenScales](https://github.com/landypro/openscales/blob/master/license.txt#L2).  That wouldn't be necessary if they could just use plain LGPL, would it?

HeikoGr, 2022-01-18:
This part is completely new for me. Thank You.

But as i read the FAQ john30 could just publish the *.o files to be fine.
There is still no need to open source the whole thing.

Don't get me wrong: i want the source to be free, too! But I think it's john30s decision and we should respect it.

mlesniew, 2022-01-18:
Great to hear we're on the same page now.  

To be honest I didn't know that only publishing the object files is sufficient to comply to the LGPL.  I assumed disclosing source is always required.

Looks like what the license aims to ensure is that anybody can build their own, potentially modified, version of the library and be still able to use it with the software that uses it.  This is easy when linking dynamically.  With static linking, relinking is required, but for that just the object files are needed.  

Looking through comments in other (closed) issues in this project it seems that so far there were no plans to release the source.  
That's a pity, but as you mentioned, opening the source code is only up to the author.