john30/ebusd-esp #66: Question: How to configure ESP for compatibility with eBUS adapter 2

muesliman, 2022-03-20:
I use a [eBUS adapter 2](https://ebus.github.io/adapter/base.en) and try to connect it with a nodeMCU (because the D1 stopped working). 
Should the adapter3 FW work with the old boards?
I tried the modes "direct RX/TX" and "swapped D7/D8". 
Then I ran the ebusd in a docker env. For the RX part only the swapped option seems to work, but the TX doesn't send anything on the bus.
Then I passed the  --generatesyn option into the container and stopped the syncer on the but. The log tells me, that an ">aa" was send, but I also receive an "unable to receive sent AUTO-SYN symbol: ERR: read timeout" error.
Also I can't see any SYNC char on the bus nor on the UART.

Can you help?

john30, 2022-03-26:
yes, the firmware is backwards compatible. you just need to configure it correctly according to the connections on the adapter (depends on how the jumpers are set)

john30, 2022-04-16:
only the options listed in the UI are usable for RX/TX

meyerd, 2022-04-20:
... ah too bad, for the esp32, any IO pins can be reconfigured for one of the UARTs. But i see, that there are restrictions when using the esp8266.