john30/ebusd-esp #67: Settings lost after save&reboot (option 0 in menu) on ESP-03

meyerd, 2022-05-14:
Hi,
currently I have cobbled together a custom esp breakout board based on the esp-03 8266 chip variant. The esp01 version is booting fine (flash with `esptool --chip esp8266 -p /dev/ttyUSB0 -b 921600 write_flash 0x00000 ebus-v3_esp01.bin)

```
esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 18:fe:34:a0:64:c7
Enabling default SPI flash mode...
Chip ID: 0x00a064c7
Hard resetting via RTS pin...
```

but when i select saving the configuration and reboot, then it reboots into default mode again (all settings lost). 
The dump eeprom dumps the factory configuration. I saw that in the initial setup, if I select "direct RX/TX (GPIO3+1)", then for the intial pin setup D0 is input and D1 output (which makes sense as it is TX). D0, however is also the Flash trigger thingy and is tied to high via external pullup to get into normal boot mode. .Does this make a difference?

I will also test again with a proper power supply instead of the MCP1700 33 something (which has 500mA rating, but that could also be an issue...)

meyerd, 2022-05-14:
one more thing. If I connect to the default AP mode in the factory reset state and got to 192.168.4.1, then the esp8266 crashes

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------␊
[21:25:26:671] ␍␊
[21:25:26:671] Exception (28):␍␊
[21:25:26:671] epc1=0x40217234 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0dda0000 depc=0x00000000␍␊
[21:25:26:671] ␍␊
[21:25:26:671] >>>stack>>>␍␊
[21:25:26:671] ␍␊
[21:25:26:671] ctx: cont␍␊
[21:25:26:671] sp: 3ffff860 end: 3fffffc0 offset: 0190␍␊
[21:25:26:671] 3ffff9f0:  4024c608 0000000a 3ffffa48 4020f018  ␍␊
[21:25:26:671] 3ffffa00:  3ffe977b 0dda0000 3ffffa3c 4020f1d7  ␍␊
[21:25:26:671] 3ffffa10:  0018001f 3ffffab4 3fff0c58 4020f209  ␍␊
[21:25:26:671] 3ffffa20:  3ffe977b 3ffffab4 3fff0c58 402056b0  ␍␊
[21:25:26:671] 3ffffa30:  00000000 696c612d 00000000 00000000  ␍␊
[21:25:26:671] 3ffffa40:  69746365 00000000 7065654b 696c412d  ␍␊
[21:25:26:716] 3ffffa50:  0a006576 00000000 312e312f 00000000  ␍␊
[21:25:26:716] 3ffffa60:  3ffffb14 000001ff 3ffffd68 3fff10dc  ␍␊
[21:25:26:716] 3ffffa70:  00000000 00000010 00000020 000000c8  ␍␊
[21:25:26:716] 3ffffa80:  3ffffaf0 3fff0c58 000009e6 40205756  ␍␊
[21:25:26:716] 3ffffa90:  3fff3524 3ffffac0 3ffffd68 00000011  ␍␊
[21:25:26:716] 3ffffaa0:  3ffe975c 00000001 3ffffd68 4020f431  ␍␊
[21:25:26:716] 3ffffab0:  3fff3524 3fff30ac 0011001f 89000020  ␍␊
[21:25:26:716] 3ffffac0:  3ffffb14 00000009 3ffffb14 4020ef8c  ␍␊
[21:25:26:716] 3ffffad0:  000000c8 3ffffd50 3ffffb14 3fff3524  ␍␊
[21:25:26:716] 3ffffae0:  000000c8 3fff0c58 3ffffd50 40205833  ␍␊
[21:25:26:761] 3ffffaf0:  40213304 00000000 000003e8 00000000  ␍␊
[21:25:26:761] 3ffffb00:  00000000 3fff3524 000009e6 4020f001  ␍␊
[21:25:26:761] 3ffffb10:  00000000 74786574 6d74682f 0900006c  ␍␊
[21:25:26:761] 3ffffb20:  3ffea5c0 3ffffd50 3ffffd68 00000000  ␍␊
[21:25:26:761] 3ffffb30:  3fff0c58 3ffffd50 3ffffd68 40207872  ␍␊
[21:25:26:761] 3ffffb40:  00000000 00000000 3ffffcb0 40219a90  ␍␊
[21:25:26:761] 3ffffb50:  53554265 67697320 3a6c616e 206f6e20  ␍␊
[21:25:26:761] 3ffffb60:  6e676973 000a6c61 65000a00 2f223d66  ␍␊
[21:25:26:761] 3ffffb70:  7265763f 65736f62 73653e22 3c313070  ␍␊
[21:25:26:761] 3ffffb80:  3c3e612f 0a3e7262 00000000 00000000  ␍␊
[21:25:26:806] 3ffffb90:  40001da0 0000000a 4021550d 3ffffcb0  ␍␊
[21:25:26:806] 3ffffba0:  3ffe90b2 00000000 3ffffc00 4021563c  ␍␊
[21:25:26:806] 3ffffbb0:  00000001 3ffffc71 00000002 3ffe8368  ␍␊
[21:25:26:806] 3ffffbc0:  00000001 3ffe8368 402199cc 40219a90  ␍␊
[21:25:26:806] 3ffffbd0:  00000000 3ffe913c 00000001 00000000  ␍␊
[21:25:26:806] 3ffffbe0:  00000002 3ffffc28 3fff1548 3ffe913f  ␍␊
[21:25:26:806] 3ffffbf0:  3ffe913e 00000000 3ffffcb0 40219ee1  ␍␊
[21:25:26:806] 3ffffc00:  00000000 00000000 0000001f 401005d0  ␍␊
[21:25:26:806] 3ffffc10:  00000001 00000004 3fffc228 401064e9  ␍␊
[21:25:26:850] 3ffffc20:  4000050c 00000000 3ffffce0 40219c91  ␍␊
[21:25:26:850] 3ffffc30:  40211302 00000030 00000010 ffffffff  ␍␊
[21:25:26:850] 3ffffc40:  0000034a 0000034a 3ffe860c 3fff3384  ␍␊
[21:25:26:850] 3ffffc50:  00000000 00000010 00000020 40101060  ␍␊
[21:25:26:850] 3ffffc60:  00000000 00000010 3ffffd10 40101098  ␍␊
[21:25:26:850] 3ffffc70:  00000020 00000020 3ffffd50 40101098  ␍␊
[21:25:26:850] 3ffffc80:  00000345 00000000 00000000 00000025  ␍␊
[21:25:26:850] 3ffffc90:  000002ff 000002ff 3ffe860c 40100e57  ␍␊
[21:25:26:850] 3ffffca0:  007a1200 957c359c 4b020c00 008d979e  ␍␊
[21:25:26:850] 3ffffcb0:  00000000 00000000 4bc6a7f0 00000000  ␍␊
[21:25:26:896] 3ffffcc0:  00000000 00000000 00000000 ffffffff  ␍␊
[21:25:26:896] 3ffffcd0:  00007a75 00000430 00000000 4020fff4  ␍␊
[21:25:26:896] 3ffffce0:  3fff0c58 ffffffff ffffffff 40210972  ␍␊
[21:25:26:896] 3ffffcf0:  00007a75 00000430 3fff3074 4020be3e  ␍␊
[21:25:26:896] 3ffffd00:  0000012c 0000033d 3ffe860c 40100e57  ␍␊
[21:25:26:896] 3ffffd10:  0000000a 3fff0c7c 402128ac 00000000  ␍␊
[21:25:26:896] 3ffffd20:  3fff0c7c 00000000 3fff0c58 4020c381  ␍␊
[21:25:26:896] 3ffffd30:  3fff0c7c 00000000 3ffffd78 4020c394  ␍␊
[21:25:26:896] 3ffffd40:  3fff0c7c 00000000 3fff0c58 40205110  ␍␊
[21:25:26:896] 3ffffd50:  74786574 6d74682f 0900006c 3fff0f64  ␍␊
[21:25:26:940] 3ffffd60:  00000013 3ffe95b0 3fff3524 09e6100f  ␍␊
[21:25:26:940] 3ffffd70:  80000000 00000000 00000000 4020f6b0  ␍␊
[21:25:26:940] 3ffffd80:  00000000 6f792072 63207275 3fff10dc  ␍␊
[... 15 lines cut ...]
[21:25:26:985] 3ffffe80:  00000001 3fff0ed0 3fff0ecc 4020855c  ␍␊
[21:25:26:985] 3ffffe90:  3ffffee6 00000008 3fff0f64 40217169  ␍␊
[21:25:27:030] 3ffffea0:  3fffff50 3fffff20 00000008 3fff0f64  ␍␊
[21:25:27:030] 3ffffeb0:  3fffff20 3fff2e84 0000002f 80000000  ␍␊
[21:25:27:030] 3ffffec0:  65746e45 676e6972 6e6f6320 4020e4dc  ␍␊
[21:25:27:030] 3ffffed0:  74736f48 656d616e 6265203a 612d7375  ␍␊
[21:25:27:030] 3ffffee0:  63343630 31000a37 3fff000a 3fff10dc  ␍␊
[21:25:27:030] 3ffffef0:  00000001 3fff0ed0 3fff0ef4 4020909c  ␍␊
[21:25:27:030] 3fffff00:  00000000 40212878 3fff0c58 40207e61  ␍␊
[21:25:27:030] 3fffff10:  3fffff50 3fffff20 00000008 3fff10dc  ␍␊
[21:25:27:030] 3fffff20:  0000029d 0000029d 3ffe860c 40100e57  ␍␊
[21:25:27:030] 3fffff30:  00000000 3fff1ab8 3fffff70 3fff10dc  ␍␊
[21:25:27:075] 3fffff40:  3fffdad0 00000020 3fff2e84 40101036  ␍␊
[21:25:27:075] 3fffff50:  4020e303 00000001 3fffff70 4020edc9  ␍␊
[21:25:27:075] 3fffff60:  3fffdad0 00000001 3fff0f64 40208109  ␍␊
[21:25:27:075] 3fffff70:  00000000 000b000f 00000000 402135d4  ␍␊
[21:25:27:075] 3fffff80:  0104a8c0 feefeffe feefeffe feefeffe  ␍␊
[21:25:27:075] 3fffff90:  feefeffe feefeffe feefeffe 3fff10dc  ␍␊
[21:25:27:075] 3fffffa0:  3fffdad0 00000000 3fff10c8 40210114  ␍␊
[21:25:27:075] 3fffffb0:  feefeffe feefeffe 3ffe8608 401013bd  ␍␊
[21:25:27:075] <<<stack<<<␍␊
[21:25:27:075] ␊
[21:25:27:075] --------------- CUT HERE FOR EXCEPTION DECODER ---------------␊
```
but that might also be related to the power supply.

meyerd, 2022-05-15:
Did just try with a proper lab-power supply and has the same behavior. So doesn't seem to be a power related issue.

salvq, 2022-06-13:
I am getting same behavior...after I set Wi-Fi AP / password etc, it is always reseted back to default values.

I tried web based config, also termanil below, same results and can not connect to WIFI AP

```

0. Save configuration and reboot
Enter your choice: 0
Configuration stored, rebooting...

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v000609a0
~ld

Welcome to eBUS adapter 3, build 20220327
Configured as WIFI access point EBUS without password.
For configuration with web browser, connect to this WIFI and open http://192.168.4.1/
Entering configuration mode.
Chip ID: 002502ac, d1mini_pro_16m
Hostname: ebus-2502ac
Configuration (new):
1. WIFI SSID: EBUS
2. WIFI secret:
3. WIFI IP address: DHCP (failed: disconnected, reason: 1)

```

salvq, 2022-06-14:
After reading thread https://github.com/john30/ebusd-esp/issues/46#issuecomment-1008057823, I can confirm that using [esphome-flasher 1.4.0](https://github.com/esphome/esphome-flasher/releases/download/1.4.0/ESPHome-Flasher-1.4.0-Windows-x64.exe) with [ebus-v3_d1mini_pro_16m.bin](https://github.com/john30/ebusd-esp/blob/master/dist/ebus-v3_d1mini_pro_16m.bin) on the Wemos D1 mini Pro 16MB works just fine, settings are saved and keep / persist after device reboot.

Using same device and firmware with NodeMCU Flasher or esptool did not work and settings are not kept after device restart

meyerd, 2022-06-15:
thanks for the hints, I also tried using the esphome flasher 1.4.0, but for the esp-03 module with 512K flash memory I get the same fault as before and the settings are not saved.

```
Using '/dev/ttyUSB0' as serial port.
Connecting....

Chip Info:
 - Chip Family: ESP8266
 - Chip Model: ESP8266EX
 - Chip ID: 00A064C7
 - MAC Address: 18:FE:34:A0:64:C7
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 512KB
 - Flash Mode: qio
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 1.0s
Flash will be erased from 0x00000000 to 0x00056fff...
Flash params set to 0x0000
Compressed 356208 bytes to 255082...
Wrote 356208 bytes (255082 compressed) at 0x00000000 in 5.8 seconds (effective 490.6 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!
```

maybe if the flash memory is "too big" (16m in your case), then it doesn't matter because the flash layout just has enough space and just some of it is unused, but in my case with a flash memory "too small", then the layout of where the settings are stored are no longer correct?

salvq, 2022-06-16:
Well, most likely it is flash size related in your case, description clearly says 1MB therefore I suggest you to get 1MB version (there is no version with 512KB flash size either).

meyerd, 2022-07-29:
@john30 is it possible that you recompile the esp01 version with a new memory layout setting for esp8266 chips with only 512k flash?

meyerd, 2022-08-30:
@john30, the latest binary release seemed to fix the issue with using the esp01 firmware for the esp03 variant of the board.

closed 2022-08-30 by meyerd