john30/ebusd-esp #69: esp d1 mini stop working when plugged in to ebus adapter v3

frompani, 2022-06-27:
hi , 
I flashed ebus-v3_d1mini.bin on esp8266 -12f , and setup wifi with no issue. 
But when I plug the esp into the adapter , esp does not connect to wifi .

Power cord and supply are good  .
If I plug power into the adapter's usb nothing happen .
If I plug power into esp's usb , adapter's blue led start blinking ,green and orange are on,  esp apparently not working .

any suggestion? 
thnak's
fab

john30, 2022-09-17:
mostly power source related, so try another one first.
and check if latest firmware is better as the WIFI signal strength was increased

frompani, 2022-09-24:
already changed 3 .
changed also esp ( twice) ... no luck .
I'm trying to use a spare RPI .
Eventually , could you sell a complete working system ( ebus+esp module )?
thanks
f

john30, 2022-09-24:
did you do some fancy port setup on the esp through the command line interface or the web?

it works fine, I'm using a wifi variant myself, so the problem is elsewhere. 

> If I plug power into the adapter's usb nothing happen .

then you probably didn't do the jumpers correctly.