john30/ebusd-esp #70: Add support LAN8720 ?

xyzroe, 2022-07-07:
Greetings!
First of all, thanks for your great work! This deserves great respect.

I would like to assemble a gateway based on ESP32 but using a wired LAN connection, - the boiler is installed in the basement, poor radio connection. 
Now there are many boards based on ESP32-WROOM (4 MB) with built-in LAN8720A (Ethernet PHY), there are even boards with POE.
Only a small modification of the existing firmware is needed.

Perhaps you will be able to implement it?
If suddenly you do not have time, but you are ready to accept help - I could, of course, under full non-disclosure terms, make this functionality possible.

john30, 2022-09-17:
for the time being I wouldn't include it as the effort is wuite high with little benefit coming from it ([adapter with Ethernet exists already)](https://adapter.ebusd.eu/index.en.html#ethernet). Plus, for reliable quick Ethernet connection there is no need for this special low-latency firmware anyway

andreiandreiandreiandreiandrei, 2023-04-26:
Hi john30,

Unfortunately W5500 provides no POE. As LAN8720 is rising with POE feature, it would be a huge improvement to ebus-esp project.

Please reconsider to include LAN8720 support.

Regards

john30, 2023-06-15:
you can use a simple adapter like [this](https://www.amazon.de/DSLRKIT-Active-Splitter-Ethernet-Raspberry/dp/B01H37XQP8/ref=sr_1_5) to extract power from PoE.

closed 2023-06-15 by john30