john30/ebusd-esp #71: Problem with Firmware for wemos D1 mini pro version 20220731

JaVier-Lloret, 2022-09-09:
I have updated my "wemos D1 mini pro" to version 20220731-d1mini_pro_16m and I cannot change the "EBUS" wifi network to mine. I had to mount version 20220731-d1mini to do it.

Could the source of the problem be that my WiFi password is 32 characters long and due to some limitation such long passwords are not supported?

After the introduction of the data of the local WiFi and after the reboot the data remain the same and the AP EBUS is maintained.

It was expected that EBUS could be changed for the value of local WiFi

I look forward to your comments.

[image]
[image]

john30, 2022-09-17:
use the [flash.sh script](https://github.com/john30/ebusd-esp/blob/master/flash.sh) to check if the memory is really that big

JaVier-Lloret, 2022-09-25:
I have confirmed that my wemos is 16Mb and tried again to program it by checking the correct option in ES8266Flasher.
The result has been the same. it does not allow to modify the WiFi network.

I have solved it using the TASMOTIZER v1.2 programmer. Once the wemos is recorded, it allows the modification of WiFi without problems.

john30, 2022-10-03:
what ES8266Flasher reveals might be wrong, I'd only trust the esptool.py output for now.
best is to use the flash.sh script which does check the memory before flashing