john30/ebusd-esp #74: Wemos D32 Pro Support

ndfred, 2022-11-05:
I have been flashing the `ebus-v3_lolin32_factory.bin` D32 firmware onto my [D32 Pro](https://www.wemos.cc/en/latest/d32/d32_pro.html), which has 16MB of flash instead of 4MB (`ESP32-WROVER` rather than `ESP32-WROOM-32`), and am running into a boot loop:

> rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:QIO, clock div:2
> load:0x3fff0030,len:1184
> load:0xffffffff,len:-1

I know the D32 Pro is not officially supported, but I am wondering if there is a way to make it work given both boards are so similar. I tried flashing `tasmota32.factory.bin` and that works no problem. I thought I would rebuild the image for my own board, but the source code is not available.

Here is the flashing procedure I have used on macOS (`screen` command commented, that is what I use to identify the boot loop):

```
#!/bin/sh -e

cd "`dirname $0`"
MODULES="esptool"
VENV_PATH="$PWD/venv"
SENTINEL_PATH="$VENV_PATH/.setupdone"
FIRMWARE="ebus-v3_lolin32_factory.bin"

if [ -f "$SENTINEL_PATH" ] && [[ $(< "$SENTINEL_PATH") == "$MODULES" ]]
then
  . "$VENV_PATH/bin/activate"
else
  echo "Creating virtual environment"
  rm -rf "$VENV_PATH"
  python3 -m venv "$VENV_PATH"
  . "$VENV_PATH/bin/activate"
  pip install --upgrade pip
  pip install $MODULES
  echo "$MODULES" > "$SENTINEL_PATH"
fi

if [ ! -f "$FIRMWARE" ]
then
  curl -L -O "https://github.com/john30/ebusd-esp/raw/master/dist/$FIRMWARE"
fi

esptool.py --port /dev/tty.usbserial-1330 erase_flash
esptool.py --port /dev/tty.usbserial-1330 write_flash 0x0 "$FIRMWARE"
# screen /dev/cu.usbserial-1330 115200
```

I have also looked at `[flash.sh](https://github.com/john30/ebusd-esp/blob/master/flash.sh)` that I modified to reuse `esptool.py` in the venv set up above, and that gives me a flash error mismatch (16MB rather than 4MB). Since the size is larger I reckon it shouldn't be a problem, so I just commented that test and got this command line, which I flashed my board with, again without success (boot loop):

> esptool.py --chip=esp32 --baud 2000000 --after hard_reset write_flash --erase-all --flash_size 4MB --compress 0 ebus-v3_lolin32_factory.bin

`ebus-v3_lolin32_factory.bin` is 946112 bytes which matches the file on the repo. Any suggestions on how to make this work? Thanks!

ndfred, 2022-11-05:
Tried with `--flash_size 16MB` and that doesn't solve it, and tried with the latest `ebus-v3_lolin32_factory.bin` (size 948960 bytes), no luck with that either.

ndfred, 2022-11-06:
It seems obvious in retrospect, but I didn't realise you needed the eBUS Adapter to deal with 24V, and the board [only accommodates the D1 mini format](https://adapter.ebusd.eu/index.en.html#wifi) directly. So if I don't want a mess of wires, that board would work best.

john30, 2022-11-10:
you can also use a D1 mini 32. anyway, the flash size needs to match of course

closed 2022-11-10 by john30