john30/ebusd-esp #76: ESP32 HW-665 image

nyxnyx, 2022-12-07:
Hi,
I've installed image (factory one since it was first install of image) for ESP32 but it didn't run. 
My chip is probably a clone of WROOM-32:
Chip id is ESP32D0WDQ6 (revision 1)
Flash size is 4MB

can someone help?

Filippo125, 2022-12-30:
Hi, 
same issue with chip id  ESP32-D0WD-V3

CherAlban, 2023-07-21:
Hi,
I am having the same issue trying to flash 20221215 on an "ESP32-D0WD-V3 (revision 3)" on a D1 clone board, using NodeMCU PyFlasher. The serial port repeats:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:2
load:0x3fff0030,len:1184
load:0x000002c6,len:738469239
ets Jul 29 2019 12:21:46

Flashing ESPEasy works just fine.

john30, 2023-08-15:
the chip is not the only thing that matters. also the IO association might be different or just not working. so, no other than the variants mentioned in the readme are supported