john30/ebusd-esp #80: Can't upload fimware

yaploxoigolyb, 2023-01-14:
Hello. I have a board on an ESP-WROOM-32 chip, I tried to flash all 4 firmware versions for this chip, all to no avail. At first I tried your script, it refused to work at all even with administrator rights and all installed packages in WSL. Then by what was in it relative to my chip. To no avail. The Flash download tool also didn't help in any way. I found a similar problem, as from the logs below, I tried NodeMCU-PyFlasher. Also to no avail.
Here is the loggers output:
- For factory images:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:2
load:0x3fff0030,len:1184
load:0x000002c6,len:738469239
(Repeats)

- For standard images:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x77434c63
invalid header: 0x77434c63
invalid header: 0x77434c63
invalid header: 0x77434c63
invalid header: 0x77434c63
invalid header: 0x77434c63
invalid header: 0x77434c63
ets Jul 29 2019 12:21:46
(Repeats)

The size of the flash drive is 4 MB. I tried it on other boards with the same chip, the problem is not solved.

juha63, 2023-02-04:
Same problem here. Board named "Mini D1 ESP32":

Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:b6:1f:35:0a:f4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 2000000
Changed.
Manufacturer: 5e
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...

After flashing with D1mini32_factory.bin:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)                                                                    
configsip: 0, SPIWP:0xee                                                                                                       
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00                                                        
mode:QIO, clock div:2                                                                                                          
load:0x3fff0030,len:1184                                                                                                       
load:0x000002c6,len:738469239                                                                                                  
ets Jul 29 2019 12:21:46

ReliableBob, 2023-03-10:
I have the same problem.
There is an ESP 32 Wrover board
I flashed it first ebus-v3_lolin32_factory.bin and then ebus-v3_lolin32.bin.
Here is the result:

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
invalid header: 0x77434c63
invalid header: 0x77434c63
invalid header: 0x77434c63
invalid header: 0x77434c63
invalid header: 0x77434c63
invalid header: 0x77434c63
invalid header: 0x77434c63
ets Jun 29, 2019 12:21:46 PM
```

The same trouble with the files ebus-v3_d1mini32_factory.bin and ebus-v3_d1mini32.bin

john30, 2023-06-15:
only the boards listed are supported

closed 2023-06-15 by john30